Avnu/OpenAvnu #666: gPTP BMCA participation Issue

KoyelHasan, 2017-08-03:
Hi,
I am using simple scenario Slave/Master[Test Station]------OpenAvnu[i210]. Found that OpenAvnu is not participating on BMCA and no announce message from OpenAvnu. Below are the details. Can you please help me, if I am missing something or this is new collateral on this branch.

**Branch**: open-avb-next, Merged Pull request #663 

**Test1**:  Slave[TS]----OpenAvnu
Running daemon: sudo ./daemon_cl ethts1
Ouptut:
INFO     : GPTP [14:10:27:163] gPTP starting
ERROR    : GPTP [14:10:27:164] Using clock device: /dev/ptp0
STATUS   : GPTP [14:10:27:164] Starting PDelay
STATUS   : GPTP [14:10:27:164] Link Speed: 1000000 kb/sec
STATUS   : GPTP [14:10:27:198] AsCapable: Enabled
**Observed:** Found that both are calculating path delay successfully, but not participating on BMCA. Only the master can participate on BMCA. So, is the default role of openAvnu in the above case is slave.? 

**Test2**: Master[TS]--OPenAvnu[Master]
Running daemon: sudo ./daemon_cl enp7s0 -T
Output:
INFO     : GPTP [14:34:51:090] gPTP starting
ERROR    : GPTP [14:34:51:091] Using clock device: /dev/ptp0
STATUS   : GPTP [14:34:51:091] Switching to Master
STATUS   : GPTP [14:34:51:091] Link Speed: 1000000 kb/sec
STATUS   : GPTP [14:34:51:140] AsCapable: Enabled
STATUS   : GPTP [14:34:52:030] Switching to Slave
STATUS   : GPTP [14:34:52:030] New Grandmaster "00:12:01:FF:FE:00:00:01" (previous "00:00:00:00:00:00:00:00")
**Observed:**  No announce message from OpenAvnu. Only TS is sending and become GM. 

Thanks in advance. Please let me know if you need any data.

Regards,
KH

andrew-elder, 2017-09-21:
Has anyone else duplicated this?
Does it still occur?

merrilldav, 2017-09-21:
Hi,
I am currently investigating this behavior with the master branch, using cda79716cd18ec755f5187c45719c5e6c6d2e563. I suspect the issue may have started with this commit:
https://github.com/AVnu/OpenAvnu/commit/27182b036e83309e7bf6eb33ac49289aa020f537

Both devices will report that they are asCapable. One device will report that it is entering slave mode, but the other does not report entering master mode (as KoyelHasan showed). Interestingly, if I swap the slave with a daemon_cl build from prior to the above commit, everything works. If I then swap the latest build back in for the slave, it still works. It appears to work fine after that until I reboot, when the issue returns using the latest build.

andrew-elder, 2017-09-21:
@merrilldav - thank you for the update!

KoyelHasan, 2017-10-03:
Hi Merrilldav,
Any update on this issue? Did you make any changes to solve this issue?

pinealservo, 2017-10-03:
@KoyelHasan You might try reverting your local repository to the commit before https://github.com/AVnu/OpenAvnu/commit/27182b036e83309e7bf6eb33ac49289aa020f537

I suspect that @merrildav is right and that commit introduced more subtle problems than we were able to catch in review. I haven't had issues myself, but I have not been exercising it very thoroughly in my typical usage either.

I have been thinking about how we might prevent these kinds of regressions, or at least catch them more quickly, but I don't have answers yet. My time for development on it is very limited.

merrilldav, 2017-10-03:
@KoyelHasan 
One of my colleagues is looking into this issue. She tells me that she sees the issue much less often with the latest build on the open-avb-next branch. Previously, we saw it every time we rebooted the devices. With the latest changes, she sees it on maybe 1 in 20 reboots. She hasn't root-caused the issue yet, so we don't know why.

Let us know if you learn anything. Thanks!

andrew-elder, 2017-10-06:
It might be helpful to post setup conditions to reproduce this issue along with working and not working logging output. That way, anyone else wishing to jump in and assist with the debugging effort has a head start on the issue.

sarh2o, 2019-06-11:
Hi,
Does anyone know whether this issue is fixed?
I tried on the latest master branch. BMCA still doesn't work.

Thanks!