Avnu/OpenAvnu #667: simple_talker running error

KoyelHasan, 2017-08-03:
Hi,
I am using Branch: open-avb-next. Merged Pull request #663. While running, simple_talker, it showing below error message. It was working fine and suddenly it stop working. Anyone have any idea about the below error message why it is happening?

@sudo ./simple_talker  -i entts1 -t 2 
@attaching to 0000:07:00.0
attach failed! (Unknown error -6)
connect failed (No such file or directory) - are you running as root?

pinealservo, 2017-08-03:
Did you perhaps reboot and forget to switch back to the OpenAvnu igb kernel driver? The "attach" mentioned is related to mapping the i210's PCI register space into the process.

KoyelHasan, 2017-08-04:
Yes, you are right. It was due to igb kerner driver. Thank you !!

closed 2017-08-04 by andrew-elder