Avnu/OpenAvnu #668: [Auto]MEDIA_READY Implementation

KoyelHasan, 2017-08-04:
Hi,
I was running the automotive profile and found that it was correctly sending AVB Ethernet and AVB Syns status message as per expected behaviour. But did not find any Media ready status message. It should  transmits the Media Ready status message when it is ready to transmit streams(in case of talker). 

Can anyone please confirm if it is implemented or not. I was browsing through the code  but did not find the same. (STATION_STATE_AVB_MEDIA_READY).

Thanks,
Koyel

pinealservo, 2017-08-08:
The gPTP daemon doesn't have the information required to determine the AVB Media Ready state. This information would have to come from a media-handling process, such as avtp_pipeline, but that has not yet been implemented.

KoyelHasan, 2017-08-09:
Thank you for your confirmation !!