Avnu/OpenAvnu #670: gPTP Building Issues

SUBBU3881, 2017-08-17:
Hi,
   Can anyone help me in addressing and resolving issues in building gPTP image in openAVNU. I am having PC with Ubuntu 12.04 with kernel v4.4.80. I am getting following issues while building gPTP in openAVNU source code.

root@sathish:/home/sathish/OpenAvnu-master# 
root@sathish:/home/sathish/OpenAvnu-master# make gptp
mkdir -p daemons/gptp/linux/build/ && make  subdir=daemons/gptp/linux/build/  -C daemons/gptp/linux/build/ 
make[1]: Entering directory `/home/sathish/OpenAvnu-master/daemons/gptp/linux/build'
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -std=c++0x -Wnon-virtual-dtor  -c ../../common/ptp_message.cpp -o obj/ptp_message.o
In file included from ../../common/avbts_clock.hpp:39:0,
                 from ../../common/ptp_message.cpp:35:
../../common/common_port.hpp: In member function ‘bool CommonPort::getLinkDelay(uint64_t*)’:
../../common/common_port.hpp:414:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -std=c++0x -Wnon-virtual-dtor  -c ../../common/ap_message.cpp -o obj/ap_message.o
In file included from ../../common/avbts_clock.hpp:39:0,
                 from ../../common/ap_message.cpp:27:
../../common/common_port.hpp: In member function ‘bool CommonPort::getLinkDelay(uint64_t*)’:
../../common/common_port.hpp:414:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -std=c++0x -Wnon-virtual-dtor  -c ../../common/avbts_osnet.cpp -o obj/avbts_osnet.o
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -std=c++0x -Wnon-virtual-dtor  -c ../../common/ether_port.cpp -o obj/ether_port.o
In file included from ../../common/ether_port.hpp:53:0,
                 from ../../common/ether_port.cpp:36:
../../common/common_port.hpp: In member function ‘bool CommonPort::getLinkDelay(uint64_t*)’:
../../common/common_port.hpp:414:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -std=c++0x -Wnon-virtual-dtor  -c ../../common/common_port.cpp -o obj/common_port.o
In file included from ../../common/common_port.cpp:35:0:
../../common/common_port.hpp: In member function ‘bool CommonPort::getLinkDelay(uint64_t*)’:
../../common/common_port.hpp:414:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -std=c++0x -Wnon-virtual-dtor  -c ../../common/ieee1588clock.cpp -o obj/ieee1588clock.o
In file included from ../../common/avbts_clock.hpp:39:0,
                 from ../../common/ieee1588clock.cpp:36:
../../common/common_port.hpp: In member function ‘bool CommonPort::getLinkDelay(uint64_t*)’:
../../common/common_port.hpp:414:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -std=c++0x -Wnon-virtual-dtor  -c ../src/linux_hal_common.cpp -o obj/linux_hal_common.o
In file included from ../../common/ether_port.hpp:53:0,
                 from ../../common/avbts_osipc.hpp:39,
                 from ../src/linux_hal_common.hpp:45,
                 from ../src/linux_hal_common.cpp:34:
../../common/common_port.hpp: In member function ‘bool CommonPort::getLinkDelay(uint64_t*)’:
../../common/common_port.hpp:414:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -std=c++0x -Wnon-virtual-dtor  -c ../src/linux_hal_persist_file.cpp -o obj/linux_hal_persist_file.o
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -std=c++0x -Wnon-virtual-dtor  -c ../../common/gptp_log.cpp -o obj/gptp_log.o
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -std=c++0x -Wnon-virtual-dtor  -c ../src/platform.cpp -o obj/platform.o
cc -Wall -g -I. -I../../common -I../src -I/root/header/include/ -c ../../common/ini.c -o obj/ini.o
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -std=c++0x -Wnon-virtual-dtor  -c ../../common/gptp_cfg.cpp -o obj/gptp_cfg.o
In file included from ../../common/gptp_cfg.hpp:41:0,
                 from ../../common/gptp_cfg.cpp:49:
../../common/common_port.hpp: In member function ‘bool CommonPort::getLinkDelay(uint64_t*)’:
../../common/common_port.hpp:414:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -std=c++0x -Wnon-virtual-dtor  -c ../src/linux_hal_generic.cpp -o obj/linux_hal_generic.o
In file included from ../../common/ether_port.hpp:53:0,
                 from ../../common/avbts_osipc.hpp:39,
                 from ../src/linux_hal_common.hpp:45,
                 from ../src/linux_hal_generic.hpp:37,
                 from ../src/linux_hal_generic.cpp:34:
../../common/common_port.hpp: In member function ‘bool CommonPort::getLinkDelay(uint64_t*)’:
../../common/common_port.hpp:414:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
../src/linux_hal_generic.cpp: In function ‘int findPhcIndex(InterfaceLabel*)’:
../src/linux_hal_generic.cpp:172:25: error: aggregate ‘findPhcIndex(InterfaceLabel*)::ethtool_ts_info info’ has incomplete type and cannot be defined
../src/linux_hal_generic.cpp:188:13: error: ‘ETHTOOL_GET_TS_INFO’ was not declared in this scope
../src/linux_hal_generic.cpp: In member function ‘virtual bool LinuxTimestamperGeneric::HWTimestamper_gettime(Timestamp*, Timestamp*, uint32_t*, uint32_t*) const’:
../src/linux_hal_generic.cpp:463:25: error: aggregate ‘LinuxTimestamperGeneric::HWTimestamper_gettime(Timestamp*, Timestamp*, uint32_t*, uint32_t*) const::ptp_sys_offset offset’ has incomplete type and cannot be defined
../src/linux_hal_generic.cpp:466:22: error: ‘PTP_MAX_SAMPLES’ was not declared in this scope
../src/linux_hal_generic.cpp:467:22: error: ‘PTP_SYS_OFFSET’ was not declared in this scope
../src/linux_hal_generic.cpp: In function ‘int findPhcIndex(InterfaceLabel*)’:
../src/linux_hal_generic.cpp:200:1: warning: control reaches end of non-void function [-Wreturn-type]
make[1]: *** [obj/linux_hal_generic.o] Error 1
make[1]: Leaving directory `/home/sathish/OpenAvnu-master/daemons/gptp/linux/build'
make: *** [gptp] Error 2
root@sathish:/home/sathish/OpenAvnu-master# 
root@sathish:/home/sathish/OpenAvnu-master# 

Please, Help me resolving this issue as soon as possible.

Thanks and Regards,
Narreddi

andrew-elder, 2017-08-17:
What have you done so far to try to solve this issue?

SUBBU3881, 2017-08-18:
Hello Andrew-Elder,
         I googled for the errors what I seen in the log, and someone said that might be issue in fetching headerfiles. Actually I am not getting any idea over these errors, because "ETHTOOL_GET_TS_INFO" macro is present in the path "/usr/src/linux-headers-3.13.0-32-generic/include/uapi/linux/ethtool.h" but our header files get fetched from the path "/usr/src/linux-headers-3.13.0-32-generic/include/linux/ethtool.h" but this ethtool.h isn't having the macro "ETHTOOL_GET_TS_INFO".

Like this I searched for this issue.

Thanks and Regards,
Narreddi

pinealservo, 2017-08-18:
Is there any particular reason you have to use Ubuntu 12.04? It is a lot easier to build in a more recent system.  In any case, since we were building with 12.04 on Travis until recently, we have the necessary hacks encoded in the .travis.yml file to get the necessary headers where the compiler will find them. As you can see by looking at that file, we just copied the necessary header files to /usr/include/linux.

SUBBU3881, 2017-08-21:
I Solved this issue by using the "include/uapi/linux/ethtool.h" along with "include/linux/ethtool.h" in the code (I added header file in .ccp file)

closed 2017-08-22 by SUBBU3881