Avnu/OpenAvnu #680: How to test gptp

SUBBU3881, 2017-08-23:
Hi,
    Can anyone help me in testing gPTP.? I am using two PCs both have installed with Ubuntu 12.04 and kernel is v4.4.80. both having Intel Network cards. As a part of my project, first I want to test OpenAVB gPTP. I tested in following way..

1. # make clean
2. # make gptp
3. # cd ../../build/obj
4. # ./daemon_cl eth0

These steps followed in both PCs. I changed time in one PC to check gPTP will change the another PC's time, but it is not happening.

Please suggest me better way of running procedure of OpenAVB gptp.

Thanks and Regards,
Siva Prakash Reddy N

andrew-elder, 2017-08-23:
How did you change the time in one PC?

Are both PCs reporting asCapable?

SUBBU3881, 2017-08-23:
Yes both are reporting asCapable.
if gptp is running in both PC's, they are showing asCapable: enabled
if gptp is running one PC it is showing asCapable: disabled

SUBBU3881, 2017-08-23:
Please mention the procedure of building and testing OpenAVB gptp

andrew-elder, 2017-08-23:
And, part one of my question?

SUBBU3881, 2017-08-23:
I am changing time using the following command..

# date -s "<date> <time>"

andrew-elder, 2017-08-23:
I don't think that will change gPTP time at all....

SUBBU3881, 2017-08-23:
Actually I am not getting how to test gPTP. Please suggest a way to test gptp

SUBBU3881, 2017-08-23:
root@prakash:/home/prakash/Open_AVB_Automotive_Profile/daemons/gptp/linux/build/obj# ./daemon_cl eth0
Using clock device: /dev/ptp0
Starting PDelay
*** Announce Timeout Expired - Becoming Master
New Grandmaster "48:4D:7E:FF:FE:E6:AF:2B" (previous "00:00:00:00:00:00:00:00")
AsCapable: Enabled

the above is actual output of gptp in one of my PC

SUBBU3881, 2017-08-23:
Please, respond

andrew-elder, 2017-08-23:
I'm volunteering my time here. I don't have complete set of test procedures at my finger tips.

bdthomsen, 2017-08-23:
As far as I can tell from the output you provided, gPTP is working correctly.  However, the gPTP code does not currently have a way to test that synchronization, and people using it have tested their synchronization in proprietary ways.  I have seen discussions regarding ways to test synchronization over the network in the IEEE 1588 and Avnu Alliance Testability groups.  You are encouraged to participate in those groups, and to contribute any solutions you implement to OpenAvnu.