Avnu/OpenAvnu #684: How the talker can know the PTS value on AVTP ?

JohnRama, 2017-08-28:
Hi AVB/TSN experts, 

Please let me ask a question regarding AVTP.
Question is how the talker can know the appropriate PTS value ?

MSRP message with "Talker advertise" attribute has the accumulated latency field, which will have the worst case latency when the message reaches the listener. So, the listener can know the worst case delay.
However, how talker can know it. It seems there is no such field on the MSRP message with "Listener Ready".  

Maybe, I misunderstand something... Would you please someone clarify on this ?

John

andrew-elder, 2017-08-29:
The assumption is that the Talker does not need to "know" the worse case latency. If you wish to "tune" the propagation delay of active streams, that is more of a system wide operation and a controller would be used to inspect the reported delays across all active listeners and the user would then be able to select a smaller delay than the default 2 millisecond delay.

JohnRama, 2017-08-29:
Andrew, 

It makes sense.

Thanks. 

John.

closed 2017-08-30 by andrew-elder