Avnu/OpenAvnu #686: I210 controller 802.1Qbv support

LanderU, 2017-08-31:
Hi!

I would like to know if it's possible to use an I210 NIC for 802.1Qbv(TSN Time Aware Shapers).

If so, where can I find some documentation/information about how to use/configure it?

pinealservo, 2017-08-31:
From what I have heard, there is a group that is creating a Qbv implementation based on the i210 using the launch time functionality of the hardware. The only documentation I'm aware of is the Qbv specification itself along with the datasheet provided by Intel for the i210, which only describes generally how to set up the launch time functionality rather than how you might use it to achieve any particular shaping protocol.

The igb library in this repository, along with the simple_talker example, gives some example code for setting up the launch time descriptor fields. I have not thought much about it, but I think implementing Qbv would require passing *all* traffic through a scheduling layer and careful assignment of launch times. Not a trivial project, but it seems possible.

LanderU, 2017-09-01:
@pinealservo thanks for the answer!

I found some patches for igb drivers (from Henrik Austad) to use it for TSN purposes. I understand that it uses the Qav functionalities to pace out the frames so it's compatible with the Qbv schedulers. I hope that with the simple_talker example I will be able to test it.

closed 2017-10-13 by LanderU

techdevop, 2017-11-07:
@LanderU Can you please help me install the **igb drivers** patch to make the driver compatible for the **Qbv schedulers**. It would be grateful if you can share the igb driver patch

LanderU, 2017-11-07:
Hi @techdevop,

we're using this [patch](
https://lkml.org/lkml/2017/9/18/76) from Richard Cochrand. With the instructions provided you should be able to use it with other Qbv capable devices. 

Let me know if you've any problem.

techdevop, 2017-11-08:
@LanderU Thanks for the response.
How to test the multicast packets egressing from the i210 interface to be in a scheduled manner?

techdevop, 2017-11-09:
Thanks for mentioning the patch link.
Can @LanderU @pinealservo please throw some light on these parameter usage/meaning **SO_TXTIME, period_nsec, waketx_delay** which is mentioned in the [patch](https://lkml.org/lkml/2017/9/18/76).
If possible can @LanderU please share the findings/program on your side which is tested with the Qbv capable device.

pinealservo, 2017-11-09:
@techdevop The following bit from the link is what's relevant for scheduling the launch time:
~~~
	/*
	 * We specify the transmission time in the CMSG.
	 */
	if (use_so_txtime) {
		msg.msg_control = u.buf;
		msg.msg_controllen = sizeof(u.buf);
		cmsg = CMSG_FIRSTHDR(&msg);
		cmsg->cmsg_level = SOL_SOCKET;
		cmsg->cmsg_type = SO_TXTIME;
		cmsg->cmsg_len = CMSG_LEN(sizeof(__u64));
		*((__u64 *) CMSG_DATA(cmsg)) = txtime;
	}
	cnt = sendmsg(fd, &msg, 0);
~~~

The `SO_TXTIME` cmsg contains a u64 timestamp, here stored in the variable `txtime`. It is the timestamp at which the packet in `msg` should be transmitted. The patch provides that cmsg and a path for it to get to the driver and into the descriptor's launch time field.

For scheduled traffic, you need to know the cycle time for the traffic schedule and your transmit offset within the cycle. You need to submit your message, programmed with the appropriate txtime value representing the application's offset into the currently-being-programmed cycle, long enough before the launch time that no max-length frame will have a chance to start transmitting from a lower-priority queue such that it won't be finished before your launch time. The i210 hardware takes care of inhibiting lower-priority queues if they would interfere with an already-scheduled frame in a SR-class queue, but it can't do it if it doesn't have the scheduled frame descriptor yet!

techdevop, 2017-11-09:
@pinealservo Thank you for helping me out

Can @pinealservo @LanderU please clarify me on the questions below 

1) Is it possible schedule my packets in the launch time specified under the `10ms cycle time` (i have attached the image for reference)?
2) If question 1 is possible, Can you please help me to set the launch time (`5ms-7.5ms period`) under `1ms cycle time`
3) Is it possible to probe the from the i210 NIC to view the packets in the scheduled interval in the oscilloscope (DSO)

`Ref: Image`
[image]

pinealservo, 2017-11-09:
1. Yes, if you can make some sense out of the diagram. I am not sure exactly what it means.
2. You will have to do these calculations yourself.
3. Yes, given a good enough DSO. You will have to figure out how yourself, or ask someone locally who can guide you.

forkmaster7, 2017-12-14:
💯 
Finally i found the right place to ask for help.

Hi @LanderU @techdevop ,
Which version of the kernel to use as the base to apply the patch mentioned in [link](https://lkml.org/lkml/2017/9/18/76).

Is RT Patch required on it? if yes, which version to apply?

LanderU, 2017-12-14:
Hi @forkmaster7 ,

the PREEMPT-RT patches are not required. But, if you want to test it using the same kernel, here you have the links:

* [Kernel](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/?h=linux-4.9.y&id=efcfbfb1d8bf756d1b58fe215bf4e419d176435b) 
* [PREEMPT-RT patches](https://www.kernel.org/pub/linux/kernel/projects/rt/4.9/older/patches-4.9.40-rt30.tar.gz)

forkmaster7, 2017-12-14:
Hi @LanderU , 
Thanks for the quick response.

I have followed the same kernel version which you have specified (**linux-4.9.40 kernel** <- **patch-4.9.40-rt30.patch** <- **[Qbv driver patch](https://lkml.org/lkml/2017/9/18/76)**)

But Im getting a build error -
[image]

Can @LanderU you please help me?

LanderU, 2017-12-14:
@forkmaster7 all the patches are applied without problems?

to check it: ` find -name "*.rej" inside of your kernel source folder.

Could you please, share how are you applying the patches?

forkmaster7, 2017-12-14:
Steps which i have followed -

1) mkdir -p ~/tmp/patch
2) cd ~/tmp/patch
3) tar -xvf linux-4.9.40.tar.gz
4) cd linux-4.9.40
5) patch -p1 < (patch files taken from this [link](https://lkml.org/lkml/2017/9/18/76))
6) cp /boot/config-$(uname -r) .config && make oldconfig
(selected the default configuration)
7) sed -rie 's/echo "\+"/#echo "\+"/' scripts/setlocalversion
8) make-kpkg clean
9) CONCURRENCY_LEVEL=$(getconf _NPROCESSORS_ONLN) fakeroot make-kpkg --initrd --revision=0 kernel_image kernel_headers

Patch is applied fine. After executing the command you have mentioned (find -name "*.rej") there was nothing printed

Can @LanderU please help with this detail?

anony2121, 2019-04-02:
I was trying to stream a song from avb talker .I had the connection been set up , I can monitor the AVTP packets been transmitted, but the audio data is null.
Can someone help me with that .

EnfangCui, 2019-10-08:
Hi,
I used the ubuntu 18.04 with Linux kernel 5.0.1 including "taprio" drivers at Networking support --> Networking options --> QOS and /or fair queueing. But when I run the "tc qdisc replace dev eth0 parent root handle 100 taprio " command, it gives error: "Unknown qdisc "taprio", hence option is unparsable". I wonder how to make the Linux support taprio. Can you give me a more detailed kernel configuration? Look forward to your kind reply. Thanks!

aguedes, 2019-10-08:
This error seems to be due to incompatible version of iproute2 package (the package 'tc' belongs to). Make sure it matches the same version of the kernel.

EnfangCui, 2019-10-10:
@aguedes  Thank you very much. I installed the latest version of iproute2 and solved the problem.