Avnu/OpenAvnu #687: Q on Clock Recovery

JohnRama, 2017-09-02:
Dear expert,

I want to understand how the media clock recovery works.
In my understanding, media clock generator uses PTS field in AVTP to tell the ticks of divided version of master clock.

1.  How the recovery device can know what division factor is used on the media clock generator and what factor should be used to multiply to recover the clock.
2.  Assuming that talker is the media clock generator. If we utilize media clock recovery,
      how the actual PTS(Presentation time stamp) can be delivered to the listener ?
    Maybe, I misunderstand something...
3. I know that CRF(Clock Recovery Format) is defined in IEEE1722-2016. Why this format is defined even though we have the mechanism using AVTP ?

John

pinealservo, 2017-09-02:
John,

Your understanding seems a little confused on some details--I think it will make more sense with a better understanding of the core concepts.

First, the timestamps in the AVTP presentation time field are *not* ticks of the media clock.  They are wall-clock ticks that correspond to when media clock edges should occur, and they are created by capturing wall-clock timestamps of when media clock edges actually did occur and then adding a fixed offset. This is a very important distinction.

By "wall-clock" I mean a high-resolution timer that is referenced to a shared time standard, i.e. gPTP Grandmaster Time. Because all nodes in the system have synchronized their measurement of wall-clock time and can reference their high-resolution timers to it, they can schedule things to happen at the same time and measure how far off things are from when they ought to be.

As to your specific questions:

1. The number of sample clock edges between timestamps is defined by the AVTP stream formats, and it varies depending on which one you're using. If you know the sample rate of the audio samples and take the difference between two presentation timestamps in a stream, it should be obvious how many clock periods that difference represents though.

2. A Talker may or may not be the media clock master; if it is, it doesn't need to do media clock *recovery*, it needs to do media clock *measurement* so that it can generate presentation time stamps. Because media clocks are required to be very stable, an endpoint with sufficient measurement capability can quickly characterize its own media clock according to its phase and frequency relationship to the wall clock and generate any number of future presentation time stamps. If it is sending stored media instead of a live capture and it is the media clock master, it can simply synthesize presentation time stamps based on the nominal sampling rate instead of measuring anything.

Any endpoint that is going to do analog/digital conversion and isn't the media clock master is going to have to do media clock recovery. Half of this is the same measurement task that the media clock master has to do; the endpoint must measure its sample clock with a high-resolution timer referenced to wall-clock time. It must then compare those timestamps for phase and frequency offset to timestamps from the clock master, deriving phase and frequency error terms. These are used to send a control signal to the local sample clock generator to adjust it towards alignment with the master clock.

3. CRF streams are useful to allow endpoints that wish to share media streams to pre-synchronize their sample clocks so that when actual media streams are started, there is no adjustment period required before the clocks are synchronized. Before CRF, people often used zero-sample media streams just to distribute the timestamp information. This was wasteful of bandwidth; CRF is much more friendly to network utilization.

JohnRama, 2017-09-05:
Levi

Thanks so much for your explanation in details.
Please see my comment below.

> John,
>
> Your understanding seems a little confused on some details--I think it will make more sense with a better understanding of the core concepts.
>
> First, the timestamps in the AVTP presentation [...]
OK. You are right. Thanks for pointing out the mistake of my description.

> By "wall-clock" I mean a high-resolution timer that is referenced to a shared time standard, i.e. gPTP Grandmaster Time. Because all nodes
> in the system have synchronized their measurement of [...]
OK. So, it represent the Presentation time stamp itself, which also means the edge of frame clock itself.
(If we are using 48KHz sampling rate, then PTS shows the edge of 48Khz clock)

> 2.
>
>     A Talker may or may not be the media clock master; if it is, it doesn't need to do media clock /recovery/, it needs to do media clock
> /measurement/ so that it can generate presentation time [...]
Got it. 

> 3. CRF streams are useful to allow endpoints that wish to share media streams to pre-synchronize their sample clocks so that when actual media
> streams are started, there is no adjustment period [...]

Thank you so much for clarifying it.
Unfortunately, still not clear to me.

So, CRF streams are also intended for deliver the sample clocks.
I can not see any difference between CRF and zero-sample media streams.
Both streams consumes bandwidth.

Does CRF is more efficient and light stream format comparing to the AAF or IEC61883-6 ?
If yes, how much ? Making more efficient is the main objective to define the CRF ?

I'm very interested in the background why CRF is newly defined.

pinealservo, 2017-09-05:
By "zero-sample" streams, I don't mean a stream with no samples in it. That's not supported by AAF or IEC61883-6. I mean a stream with meaningless samples that are discarded.

I'm not going to do the bandwidth calculations for you, but CRF is much lighter weight and much more flexible; it can be used to distribute arbitrary clocks at whatever rate is necessary without having to distribute any associated media. The recommended CRF rate for 48KHz audio is 6 timestamps per PDU and 50 PDUs per second; given that Class A media streams are typically 8000 PDUs per second, each of which would have a timestamp and 6 media sample frames, the increase in efficiency is significant.

It's just a smart way to do things if you're going to have a number of talkers and listeners that all need to dynamically establish streams that are synchronous to the same clock. This doesn't describe all situations where AVB/TSN is useful, but supporting CRF isn't required in case you're in a situation where it isn't useful.

JohnRama, 2017-09-05:
Levi

Thank you so much !!
Now, everything is very clear to me.
Really appreciated. 

John

closed 2021-01-26 by JohnRama