Avnu/OpenAvnu #711: need help running openavb_host

JXter, 2017-10-24:
Hi all,
I have been trying to run openavb_host application and I am getting the following errors.

jx@jx:~/Documents/OpenAvnu/lib/avtp_pipeline/build/bin$ sudo ./openavb_host -I eth2 echo_talker.ini
attaching to 0000:03:00.0
[1508823625:762526853 OPENAVB osalTime] INFO: local_time = 0
[1508823625:762535763 OPENAVB osalTime] INFO: ml_phoffset = 0, ls_phoffset = 0
[1508823625:762547133 OPENAVB osalTime] INFO: ml_freqffset = 0.000000, ls_freqoffset = 0.000000
[1508823625:762550691 OPENAVB Endpoint] INFO: AVTP Pipeline: 0.1.3 (Development)
[1508823625:765872520 OPENAVB Raw Socket] ERROR: Checking interface; ioctl(SIOCGIFFLAGS) failed: No such device
[1508823625:765897988 OPENAVB Endpoint] ERROR: Invalid value: section=network, name=ifname, value=eth0
[1508823625:765961590 OPENAVB Endpoint] ERROR: Error in INI file: endpoint.ini, line 4
[1508823625:766056216 OPENAVB Talker / Listener] INFO: AVTP Pipeline: 0.1.3 (Development)
[1508823625:766448767 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbMapPipeInitialize
[1508823625:766470476 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbIntfEchoInitialize
[1508823625:768631078 OPENAVB HAL Ethernet] INFO: TX buffers: 40
[1508823625:768642785 OPENAVB HAL Ethernet] INFO: IGB launch time feature is DISABLED
[1508823625:789343105 OPENAVB Endpoint] ERROR: Failed to initialize SRP
[1508823628:071974964 OPENAVB Endpoint MAAP] INFO: MAAP address range 2 acquired:  0x91e0f000a861-0x91e0f000a880 (Size 32)
[1508823628:072509590 OPENAVB Endpoint] ERROR: Make sure that mrpd daemon is started.
attaching to 0000:03:00.0
[1508823629:074083795 OPENAVB HAL Ethernet] INFO: TX buffers: 80
[1508823629:074093406 OPENAVB HAL Ethernet] INFO: IGB launch time feature is DISABLED
[1508823629:074196891 OPENAVB Endpoint] ERROR: Failed to initialize SRP
[1508823631:281262507 OPENAVB Endpoint MAAP] INFO: MAAP address range 3 acquired:  0x91e0f000d6c2-0x91e0f000d6e1 (Size 32)
[1508823631:281598239 OPENAVB Endpoint] ERROR: Make sure that mrpd daemon is started.

I am using 2 PCs with I210 NIC and without any switch.
I am running gPTP,SRP and MAAP in other terminals and openavb_host on one
eth2 is my interface with I210 and seems like application takes eth0 only

am I doing it correct? or is there anything that I should do differently?

I was able to run simple_talker, simple_listener applications successfully.
Thanks and regards

pinealservo, 2017-10-24:
The error message indicates that the setting for eth0 is coming from endpoint.ini on line 4. Try changing the ifname setting there to eth2. I'm not sure why it's not initializing SRP; are you sure the mrpd application is running?

JXter, 2017-10-24:
Thank you for the quick reply,
Changed ifname to eth2 and that solves first error but still mrpd error exists.
I am running the mrpd daemon and it stays at process_events(). I hope that is the correct log. 
If not please tell me how to correct.

My complete log is 

On System 1.

jx@jx-desktop:~/Documents/avb/OpenAvnu$ sudo ./run_gptp.sh eth1
groupadd: group 'ptp' already exists 
INFO     : GPTP [17:06:08:275] gPTP starting 
ERROR    : GPTP [17:06:08:276] Using clock device: /dev/ptp0 
STATUS   : GPTP [17:06:08:276] Starting PDelay 
STATUS   : GPTP [17:06:08:276] Link Speed: 1000000 kb/sec 
STATUS   : GPTP [17:06:08:310] AsCapable: Enabled 

jx@jx-desktop:~/Documents/avb/OpenAvnu$ sudo ./run_srp.sh eth1 
process_events()

On system 2: 

jx@jx:~/Documents/OpenAvnu$ sudo ./run_gptp.sh eth2 
groupadd: group 'ptp' already exists 
INFO     : GPTP [17:06:00:960] gPTP starting 
ERROR    : GPTP [17:06:00:960] Using clock device: /dev/ptp0 
STATUS   : GPTP [17:06:00:960] Starting PDelay 
STATUS   : GPTP [17:06:00:960] Link Speed: 1000000 kb/sec 
EXCEPTION: GPTP [17:06:03:994] PDelay Response Receipt Timeout 
STATUS   : GPTP [17:06:09:104] AsCapable: Enabled 
STATUS   : GPTP [17:06:09:732] Switching to Slave 

jx@jx-desktop:~/Documents/avb/OpenAvnu$ sudo ./run_srp.sh eth1 
process_events()

jx@jx:~/Documents/OpenAvnu$ sudo ./run_maap.sh eth2 
Enter "help" for a list of valid commands. 
[1508845000:498601119 MAAP Init] STATUS: Server started 
[1508845013:768843971 MAAP Init] INFO: New connection from 127.0.0.1 on socket 5 
[1508845013:769287618 MAAP Init] INFO: MAAP initialized:  0x91e0f0000000-0x91e0f000fdff (Size: 65024) 
[1508845013:769372268 MAAP Init] INFO: Address range 1 querying:  0x91e0f000c84f-0x91e0f000c86e (Size 32) 
[1508845015:849602929 MAAP Init] INFO: Address range 1 acquired:  0x91e0f000c84f-0x91e0f000c86e (Size 32) 
[1508845015:849763451 MAAP Init] INFO: Address range 1 released:  0x91e0f000c84f-0x91e0f000c86e (Size 32) 
[1508845015:849855982 MAAP Init] INFO: Socket 5 closed 

jx@jx:~/Documents/OpenAvnu/lib/avtp_pipeline/build/bin$ sudo ./openavb_host echo_talker.ini 
ifname=eth2 
attaching to 0000:03:00.0 
[1508845013:766040419 OPENAVB osalTime] INFO: local_time = 0 
[1508845013:766048573 OPENAVB osalTime] INFO: ml_phoffset = 0, ls_phoffset = 0 
[1508845013:766056563 OPENAVB osalTime] INFO: ml_freqffset = 0.000000, ls_freqoffset = 0.000000 
[1508845013:766059676 OPENAVB Endpoint] INFO: AVTP Pipeline: 0.1.3 (Development) 
[1508845013:766300095 OPENAVB Talker / Listener] INFO: AVTP Pipeline: 0.1.3 (Development) 
[1508845013:766433357 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbMapPipeInitialize 
[1508845013:766445877 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbIntfEchoInitialize 
[1508845013:767850299 OPENAVB HAL Ethernet] INFO: TX buffers: 40 
[1508845013:767859417 OPENAVB HAL Ethernet] INFO: IGB launch time feature is DISABLED 
[1508845013:868545868 OPENAVB Endpoint] ERROR: Failed to initialize SRP 
[1508845015:849670935 OPENAVB Endpoint MAAP] INFO: MAAP address range 1 acquired:  0x91e0f000c84f-0x91e0f000c86e (Size 32) 
[1508845015:849995234 OPENAVB Endpoint] ERROR: Make sure that mrpd daemon is started. 
attaching to 0000:03:00.0 

Thanks and regards

bdthomsen, 2017-10-24:
Just seeing "process_events()" is typical when running the srp daemon.  Try running it with "eth2" (instead of "eth1") on system 2 and see if that makes a difference.

JXter, 2017-10-25:
I changed the interface to eth2 on system 2 but still the log is same

JXter, 2017-10-27:
It seems like the SRP is failing because mrp_get_domain function fails.
While running simple_talker I had to press ctrl+c while the process hangs, just like you mentioned in Issue 479 @pinealservo 
How to fix this ?

closed 2017-10-30 by JXter