Avnu/OpenAvnu #713: OpenAVB building issue

CHM-me-ch, 2017-10-27:
Hi,

I am trying to get OpenAVB in my Ubuntu 16.04 machine. I am following the steps given in the link : https://github.com/AVnu/OpenAVnu/wiki/Getting-the-source. 
While building the same, i am getting error when i run the ./travis.sh command.
The error i am getting is given in this text file: 

[erroR.txt](https://github.com/AVnu/OpenAvnu/files/1421129/erroR.txt)

I have also installed the igb driver for intel i210 and the binary is installed in '/lib/modules/4.11.12-rt15/updates/drivers/net/ethernet/intel/igb/igb.ko'
igb driver version: igb 5.3.5.12

Also executed the following commands:
Install tpciutils : sudo apt-get install pciutils
sudo apt-get install g++
sudo apt-get install -y libpcap-dev libpci-dev libsndfile1-dev libjack-dev linux-headers-4.4.0-97-generic cmake
sudo cp /usr/src/linux-headers-4.4.0-97/include/uapi/linux/ethtool.h /usr/include/linux

Can anyone please help me to find a solution for this? Any help is highly appreciated. I am a newbie in linux, so please correct me if you find out the mistake i am doing.

Thanks in advance

andrew-elder, 2017-11-06:
Have you reviewed these steps <https://github.com/AVnu/OpenAvnu/tree/master/lib/avtp_pipeline> ?

CHM-me-ch, 2017-11-06:
No, I haven't. But i will do it now and see.
Before continuing, can you please confirm one thing?
The kernel version which i am working is : 4.11.12-rt15 , but the linux header is not installed for this. Is that an issue while building OpenAVB code?

CHM-me-ch, 2017-11-09:
I installed the PREEMPT_RT 4.9.47-rt37 and it solved the problem with the creation of igb_avb.ko file.Turns out the 4.11.12 linux RT is not compatible.
But still facing errors when trying to run the simple_talker and listener example. I am using 2 intel i210 cards in the same PC where one acts as listener and the other as talker. Can this be implemented? 

I am trying to implement the Time triggered send capabilities here, is this already implemented in OpenAVB examples?

pinealservo, 2017-11-09:
I don't believe using two of them in one PC is supported right now with the igb-avb driver. But when you run simple_talker it should program the launch time descriptors and send them at the programmed time. The avtp_pipeline code can also be built with support for the igb-avb driver, in which case it will also use the launch time descriptors of the i210.

CHM-me-ch, 2017-11-10:
Thankyou for the reply @pinealservo 
I tried to run the simple talker example but i am getting the following errors:

command: 
`cd /home/username/Open-AVB/daemons/gptp/linux/build/obj
sudo ./daemon_cl enp1s0 -R 1`

INFO     : GPTP [08:43:03:150] gPTP starting
ERROR    : GPTP [08:43:03:151] Using clock device: /dev/ptp0
STATUS   : GPTP [08:43:03:151] Starting PDelay
STATUS   : GPTP [08:43:03:151] Link Speed: 1000000 kb/sec
EXCEPTION: GPTP [08:43:06:184] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:07:199] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:08:214] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:09:229] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:10:243] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:11:258] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:12:273] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:13:288] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:14:303] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:15:317] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:16:332] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:17:347] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:18:362] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:19:377] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:20:391] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:21:406] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:22:421] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:23:435] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:24:450] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:25:465] PDelay Response Receipt Timeout
EXCEPTION: GPTP [08:43:26:479] PDelay Response Receipt Timeout

Then I did the Stream reservation daemon as follows:
`cd /home/username/Open-AVB/daemons/mrpd
 sudo ./mrpd -mvs -i enp1s0
[sudo] password for username:
process_events()
`
It gets stopped here:

#Run the simple talker example:
`cd /home/username/Open-AVB/examples/simple_talker
sudo ./simple_talker -i enp1s0 -t 2
[sudo] password for username:
attaching to 0000:01:00.0
MSRP:Empty
 unhandled from mrpd
`

What the simple_talker is not working? I have igb_avb driver in the interface. Confirmed using the ethtool -i command,.

Any suggestions or comments will be of great help.

andrew-elder, 2017-11-13:
You need to work on getting gPTP to run correctly first. It would appear from the log that the NIC that gPTP is using is not connected to a device that knows how to send a PDelay Response. I would double check your configuration.

I would also switch to using the avtp pipeline code here <https://github.com/AVnu/OpenAvnu/tree/master/lib/avtp_pipeline> for future testing. There are more people running it more regularly.

christopher-s-hall, 2017-11-13:
Adding to @andrew-elder's comment: If you have two PCs connected back-to-back you'll need to start gPTP on both.

ethan-stewart-iol, 2017-12-01:
> I am using 2 intel i210 cards in the same PC where one acts as listener and the other as talker. Can this be implemented?
It is possible to do this with avtp_pipeline, however there are several modifications you'll need to make. The big one is regarding getting the gptp times from the gptp daemon;

The gptp daemon needs to be modified to use two different shared memory locations, and the avtp_pipelines need to be directed to these.

At that point, avtp_pipeline will be able to stream across the nics in some circumstances. This is an issue with how avtp_pipeline (really the igb library) attaches to pci devices.

andrew-elder, 2017-12-01:
MRP will need some modification as well. There is a single UDP port used for commands and notifications.

sofiamorseletto, 2019-06-18:
Hello, everyone. I'm having some trouble compiling the project with the `AVB_FEATURE_IGB=1`.
The steps I'm following are:

- Cloning the [OpenAVnu ](https://github.com/AVnu/OpenAvnu) repository 

- `git submodule init`

- `git submodule update`

- Modifying `PLATFORM_TOOLCHAIN?=x86_i210_linux` in _avtp_pipeline.mk_

- Running `sudo ./travis.sh` on the root folder of the OpenAVnu repository

After doing this I get the following error:

```
In file included from /.../OpenAvnu/lib/avtp_pipeline/platform/Linux/rawsock/openavb_rawsock.c:38:0:
/.../OpenAvnu/lib/avtp_pipeline/platform/Linux/rawsock/igb_rawsock.h:36:17: fatal error: igb.h: No such file or directory compilation terminated.
```

Any idea of why I might be getting this error? Am I missing some installation or library in particular?

Any help is appreciated.
Thanks.
Best regards,
Sofia

JakubMielczarek81, 2019-06-18:
Hello Sofia,

Try this patch:

diff --git a/lib/avtp_pipeline/platform/Linux/x86_i210_linux.cmake b/lib/avtp_pipeline/platform/Linux/x86_i210_linux.cmake
index 48e62223..db5f882a 100644
--- a/lib/avtp_pipeline/platform/Linux/x86_i210_linux.cmake
+++ b/lib/avtp_pipeline/platform/Linux/x86_i210_linux.cmake
 # and another kernel sources
@@ -23,28 +25,29 @@ set ( OPENAVB_PLATFORM "${OPENAVB_HAL}-${OPENAVB_OSAL}" )
 # Platform Additions
 set ( PLATFORM_INCLUDE_DIRECTORIES
        ${CMAKE_SOURCE_DIR}/platform/x86_i210/include
-if (AVB_FEATURE_IGB)
-       ${CMAKE_SOURCE_DIR}/../igb
-endif ()
+if (${AVB_FEATURE_IGB})
+       ${CMAKE_SOURCE_DIR}/../igb_avb/lib
+endif()
        ${CMAKE_SOURCE_DIR}/openavb_common
        ${CMAKE_SOURCE_DIR}/../../daemons/common
        ${CMAKE_SOURCE_DIR}/../../daemons/mrpd
        ${CMAKE_SOURCE_DIR}/../../daemons/maap/common
 )
 
-if (AVB_FEATURE_IGB)
+if (${AVB_FEATURE_IGB})
        set ( PLATFORM_LINK_DIRECTORIES
-               ${CMAKE_SOURCE_DIR}/../igb
+               ${CMAKE_SOURCE_DIR}/../igb_avb/lib
        )
 endif ()
 
-if (AVB_FEATURE_IGB)
+if (${AVB_FEATURE_IGB})
        set ( PLATFORM_LINK_LIBRARIES
                igb
                pci
        )
 endif ()
 
+message (INFO " Dirs: ${PLATFORM_INCLUDE_DIRECTORIES}")
 # TODO_OPENAVB : need this?
 # Set platform specific define
 #set ( PLATFORM_DEFINE "AVB_DELAY_TWEAK_USEC=15" )

Best regards,
Jakub

sofiamorseletto, 2019-06-19:
Hello, Jakub. Thank you very much, with your help I could build the project!

On my _avtp_pipeline.mk_ I set:

```
    AVB_FEATURE_ENDPOINT ?= 1
    IGB_LAUNCHTIME_ENABLED ?= 1
    AVB_FEATURE_GSTREAMER ?= 0
    PLATFORM_TOOLCHAIN ?= x86_i210_linux
```

However, when I ran the following:

Talker:  `sudo ./openavb_harness mpeg2ts_file_talker.ini`

Listener:  `sudo ./openavb_harness mpeg2ts_file_listener.ini`

I get a `Segmentation Fault` error on the listener. Has this happened to you, too? Any idea how can I solve this?

Any help is appreciated.
Thank you again.
Best regards,
Sofia

JakubMielczarek81, 2019-06-24:
Hello Sofia,
I cannot recall this kind of issue, but it has been a while since I ran such configuration. Maybe you could run listener under gdb or ddd and then you would have some indication where it is crashing.
Best regards,
Jakub