Avnu/OpenAvnu #714: avtp pipeline alsa talker and listener

alierol, 2017-10-28:
I cloned and started to play with AVTP pipeline recently. I just could not make echo listener and talker work but as my real purpose was to establish two away communication between two machines. I started to play with alsa listener and talker and somehow managed to make them work (one way only) but there are two glitches that I would like to mention here. I am using Ubuntu 14.04 and got the same results with both the master and the open-avb-next branches.

1. First I need to apply the fix mentioned in issue #382 for the mrp_get_domain() function.
2.  The second one is quite strange. When I run the openavb_host with any configuration file  it gives an "Failed to initialize SRP " error even with the previous fix. As a strange fix, before running openavb_host I run "live_stream demo". It just hangs due to an "MSRP: Empty" response and after I break it goes on sending packets and then I completely kill it. After this operation everything with the alsa talker and listener just goes fine. ıt is enough to do it only once. 

Any help on how to get rid of the second glitch and establishing two way communication would be greatly appreciated.

closed 2017-11-08 by alierol

athul7pk, 2017-11-28:
Hi alierol ,
I am a beginner in ALSA and openavb .can you please help me to run alsa talker application.
(I guess we may have to make a loopback device and we have to edit asound.rc file ). 
if I am playing a media using mediaplayer how can i reroute to avb? how it works?

alierol, 2017-11-28:
There are three ways to use the alsa with openavb.

1. live_stream application in the examples folder. There are talker and listener applications that pushes or pulls data from gsreamer-1.0 pipeline. On the command line given in the README file just make sure to supply the correct alsa device name for the source and sink of the pipeline. The example works but introduces a 1-2 second delay. It should be possible to play media files by simply changing the gstreamer source to a file instead of alsa.

2.  gstreamer-avb-plugins in the examples folder. That's probably a better way to combine gstreamer and openavb but I could not make it work. It looks like an abandoned project.

3. Use the avtp_pipeline. avtp_pipeline has its own interface to alsa. Use the openavb_harness applications (See the command line samples in the README file of avtp_pipeline) together with the alsa_talker.ini and alsa_listener.ini files (in the same folder with the app.) If you have a properly configured default input and output device for alsa you just need to enter the  the mac address of the talker. (No need for the MAC address of the listener).  To play media files, I guess the best way is to use the corresponding interface of avtp_pipeline.

athul7pk, 2017-11-30:
Hi alierol,
Thank you very much for your quick and neat reply
I have tried with avtppipeline as you mentioned with default interface..then I faced poor audio quality and noise...after that I created a loopback device it's working fine for me
Thanks alot