Avnu/OpenAvnu #717: a question regarding algorithm for caculating pdelay.

jongwook00lee, 2017-11-02:
i'd like to chek out where my understanding is correct or not.

i think pdelay should be caculated latency between ptp master and slave.
but daemon_cl, as far as i know, has not a logic to filter pdelay_resp/resp_followup messages from other slave. 
if there are several ptp slaves, how can pick up the master's pdelay_resp/resp_folloup for calculating pdelay?

pinealservo, 2017-11-02:
PDelay is used to calculate the immediate link delay with the link peer, not with the GM. The bridges along the path from GM to every AS-capable endpoint pass along the information necessary to calculate the complete path delay given the PDelay-calculated delay on the final link. The packets are not forwarded, so no filtering is necessary on a point-to-point medium like Ethernet.

jongwook00lee, 2017-11-03:
thanks pinealservo, your comment is so helpful to me.
i have to try to check using end-points and switch i have.

closed 2017-11-03 by jongwook00lee