Avnu/OpenAvnu #720: Basic hardware and software requirements

sucre123, 2017-11-06:
Dear expert,

I’m a student  would like to study OpenAvnu. I could only find a little documents about OpenAvnu. Because I'm a new learner of this, so could someone please tell me the basic hardware and software requirements to do this? 
With much thanks.

Sucre.

andrew-elder, 2017-11-06:
I would choose an Intel CPU and an Intel i210 NIC as a starting point. See also <https://github.com/AVnu/OpenAvnu/tree/master/lib/avtp_pipeline>

techdevop, 2017-11-07:
@sucre123 
SW Requirement -
  UBUNTU 14.0.4
  kernel : 3.13.0-37 and i686.
  this configuration is working fine
HW Requirement - 
  NIC intel i210

sucre123, 2017-11-07:
Thanks for your help.
 OK, an intel i210 NIC and I will walk on my first step on OpenAVnu. 
Too much thanks. @andrew-elder @techdevop

closed 2017-11-07 by pinealservo