Avnu/OpenAvnu #721: simple_listener using higher CPU

vasuhere, 2017-11-07:
Hi,
simple listener is using higher CPU compared to talker (from benchmark data [here](https://github.com/AVnu/OpenAvnu/blob/master/lib/avtp_pipeline/README.md )). What is the reason?

Is it because of libpcap pooling?. I have I217-LM for listener PC. Is there an API to take packets from I217-LM driver (similar to igb_xmit to write data to I210 driver on talker side)

pinealservo, 2017-11-07:
We have not done detailed profiling related to those benchmark results, so I can't say why the listener might use more CPU than the talker.

There is no special driver for any NIC except the i210; it is the only one that supports timed launch timestamps in the descriptors.

vasuhere, 2017-11-20:
Thanks @pinealservo .

I was able to bring down the CPU loading on listener by changing the read timeout value in pcap_open_live (refereed [here](https://github.com/jvinet/knock/issues/5)).