Avnu/OpenAvnu #722: I210 NIC 802.1Qbv support driver

techdevop, 2017-11-07:
Hello,

Can anyone help me configuring **802.1Qbv time aware shaper** in **i210 NIC**. It would be nice if i get some direction on installing the driver and igb driver patch files.

pinealservo, 2017-11-07:
The i210 doesn't have a Qbv time aware shaper except in the sense that you can schedule individual frames, which could be used by another software layer to emulate the behavior specified in Qbv. We don't have any software that does that yet, and so we can't offer any help. The userspace i210 driver in this repository (igb) along with the support library (libigb) allows programming the launch time descriptors, though, so you could start with that. They should be built automatically when you do a top-level make.

closed 2017-11-07 by pinealservo

techdevop, 2017-11-08:
@pinealservo Thanks for the information. Timely help means a lot to me 👍 . 
Can you just let me know that the AVB stack running on the i210 interface can forward streams to TSN endpoint gateway?

pinealservo, 2017-11-08:
I don't know what you mean by "TSN endpoint gateway", or "forward streams" for that matter. Beyond the somewhat-complete AVB system implemented in the avtp_pipeline code, the components in this repository are something of a "built it yourself" kit.

You can use PTP and SRP and the igb/libigb as part of an endpoint for TSN use cases beyond those covered by AVB and avtp_pipeline, but we don't currently have any code to do that. I can't offer any advice or guidance without more information about what you are actually trying to do.