Avnu/OpenAvnu #723: Query regarding OpenAVB API's

CHM-me-ch, 2017-11-12:
I wanted to know more about the OpenAVB projects and as I go through it I have some queries regarding the API's and the drivers. I try executing the simple_talker example, but it's giving some errors which I have already posted on the site.As I look more into it, I have the following queries.
1. The API's used or the supporting API's in OpenAVB
2. How does the igb and igb_avb differ?What are the features of igb_avb driver?
Can anyone please help me with these questions?

CHM-me-ch, 2017-11-13:
Another question that came up is, whether the time triggered send feature of i210 driver is implemented in Open-AVB? when i go through the examples given, i am not able to spot the same. Can anyone confirm on this?

CHM-me-ch, 2017-11-15:
Any comments or sugggestions regarding the 'TTS (time triggered send) feature of I210' implemented/not implemented in Open-AVB?

pinealservo, 2017-11-16:
The igb_avb driver (i.e. the one in this repository rather than the upstream kernel igb driver) has a special interface that allows the igb library (also in this repository) to manage the SR-class queues from userspace while the kernel continues to manage best-effort traffic.

The igb library does allow setting the launch time for transmit descriptors. Look at the simple_talker example code to see how to do that.