Avnu/OpenAvnu #724: echo_talker and echo_listener issue

Joezhaoz, 2017-11-14:
Hi everyone, I am trying to run echo_talker and echo_listener on one PC which have two I210 card installed and all connected to a MOTU switch. I compiled daemons and build the avtp_pipeline with AVB_FEATURE_ENDPOINT=0 which is to use static stream reservation without SRP. After run the igb and gPTP daemon on both I210 card, I started echo_listener and echo_talker example, but I found that the listener did not receive the strings streamed from talker, below is the log printed, anyone can tell me why it does not work, or what steps I have missed? Thank you very much. 
Listener log:
/work/OpenAvnu/OpenAvnu$ sudo ./run_echo_listener.sh eth1
attaching to 0000:05:00.0
[1510629312:888337186 OPENAVB osalTime] INFO: local_time = 1510629312774278918
[1510629312:888351106 OPENAVB osalTime] INFO: ml_phoffset = 1510211531600024103, ls_phoffset = 87924470
[1510629312:888376167 OPENAVB osalTime] INFO: ml_freqffset = 1.000013, ls_freqoffset = 0.999976
[1510629312:890475245 OPENAVB HAL Ethernet] INFO: TX buffers: 40
[1510629312:890483289 OPENAVB HAL Ethernet] INFO: IGB launch time feature is DISABLED
[1510629312:890494585 OPENAVB Talker / Listener] INFO: AVTP Pipeline: 0.1.3 (Development)
[1510629312:890718630 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbMapPipeInitialize
[1510629312:890727294 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbIntfEchoInitialize
[1510629312:897718788 OPENAVB Listener] INFO: Attach a0:36:9f:97:85:1c/1
[1510629312:897723264 OPENAVB Listener] INFO: Dest Addr: 00:00:00:00:00:00
[1510629312:897725231 OPENAVB Listener] INFO: Starting stream: a0:36:9f:97:85:1c/1
[1510629312:897729171 OPENAVB Raw Socket] INFO: Using *igb* implementation
[1510629313:929028917] RX UID:1, calls=10, frames=0, lost=-1, bytes=0, rxbuf=-1, mqbuf=0, mqrdy=0
[1510629314:930592262] RX UID:1, calls=10, frames=0, lost=0, bytes=0, rxbuf=-1, mqbuf=0, mqrdy=0
[1510629315:932381322] RX UID:1, calls=10, frames=0, lost=0, bytes=0, rxbuf=-1, mqbuf=0, mqrdy=0
[1510629316:934052524] RX UID:1, calls=10, frames=0, lost=0, bytes=0, rxbuf=-1, mqbuf=0, mqrdy=0
[1510629317:935355920] RX UID:1, calls=10, frames=0, lost=0, bytes=0, rxbuf=-1, mqbuf=0, mqrdy=0
[1510629318:936743193] RX UID:1, calls=10, frames=0, lost=0, bytes=0, rxbuf=-1, mqbuf=0, mqrdy=0
[1510629319:938037705] RX UID:1, calls=10, frames=0, lost=0, bytes=0, rxbuf=-1, mqbuf=0, mqrdy=0
[1510629320:939306459] RX UID:1, calls=10, frames=0, lost=0, bytes=0, rxbuf=-1, mqbuf=0, mqrdy=0

Talker log:
/work/OpenAvnu/OpenAvnu$ sudo ./run_echo_talker.sh eth0
attaching to 0000:05:00.0
attach_tx failed! (Device or resource busy)
attaching to 0000:04:00.0
[1510629316:539329131 OPENAVB osalTime] INFO: local_time = 1510629316436564746
[1510629316:539335390 OPENAVB osalTime] INFO: ml_phoffset = 1510211531587422930, ls_phoffset = 100561666
[1510629316:539344899 OPENAVB osalTime] INFO: ml_freqffset = 1.000025, ls_freqoffset = 0.999964
[1510629316:540259524 OPENAVB HAL Ethernet] INFO: TX buffers: 40
[1510629316:540263298 OPENAVB HAL Ethernet] INFO: IGB launch time feature is DISABLED
[1510629316:540267071 OPENAVB Talker / Listener] INFO: AVTP Pipeline: 0.1.3 (Development)
[1510629316:540362559 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbMapPipeInitialize
[1510629316:540369504 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbIntfEchoInitialize
[1510629316:549957651 OPENAVB Talker] INFO: Dest Addr: 91:e0:f0:00:fe:00
[1510629316:549961901 OPENAVB Talker] INFO: Starting stream: a0:36:9f:97:85:1c/1
[1510629316:549966480 OPENAVB Raw Socket] INFO: Using *igb* implementation
[1510629316:578789539 OPENAVB Talker] INFO: a0:36:9f:97:85:1c/1, sr-rate=4000, data-rate=4000, frames=1, size=118, batch=1, sleep=250us, sr-Kbps=4352, data-Kbps=4352
[1510629317:578862836] TX UID:1, calls=4000, frames=4000, late=0, bytes=266890, txbuf=0, mqbuf=0, 
[1510629318:578868256] TX UID:1, calls=4000, frames=4000, late=0, bytes=268000, txbuf=0, mqbuf=0,

bdthomsen, 2017-11-14:
Reservations are used to tell the switch which Talkers and Listeners are connected, so it knows where to forward the multicast traffic.  As far as I know, there is no way to manually configure the MOTU switch to provide that information, so your options are limited to getting the Endpoint and MSRP support working, or using an Ethernet cable to directly connect eth0 and eth1 (i.e. not use the MOTU switch).

Joezhaoz, 2017-11-15:
Thank you Brant, it's solved.

closed 2017-11-15 by Joezhaoz

chuanqima, 2019-07-09:
Hello, Joezhaoz,
      I was lucky enough to find your discussion record. Now I am just trying to use avb, but I have a problem. I directly connected two I210 NICs, running gptp, one pc is the master, one pc is slave, but when I run the script run_echo_talker/run_echo_listener, I don't receive anything on the listener side (is it a string? Or a sound? What is the result?), here is the log on my two PCs, I look forward to you. The answer.
Tks.
on talker:
[image]

Joezhaoz, 2019-07-09:
Hi,

Listener will continously receive a string which is set in the talker ini
file.