Avnu/OpenAvnu #725: gst-plugin build error

JXter, 2017-11-16:
Hi all,
I was trying to build the gst-plugin and encountered some errors saying some files cannot be located.
I am using open-avb-next branch.

After configuring, while executing autogen.sh like
jx@jx:~/Documents/OpenAvnu/examples/gstreamer-avb-plugins/gst-plugin$./autogen.sh

I got error( I am not posting the entire log )

gstavbsink.c:60:33: fatal error: ../../../common/avb.h: No such file or directory
 #include <../../../common/avb.h>
                                 ^
compilation terminated.
make[2]: *** [libgstplugin_la-gstavbsink.lo] Error 1
make[2]: Leaving directory `/home/jx/Documents/OpenAvnu/examples/gstreamer-avb-plugins/gst-plugin/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/home/jx/Documents/OpenAvnu/examples/gstreamer-avb-plugins/gst-plugin'
make: *** [all] Error 2
autoreconf: make failed with exit status: 2
autogen.sh failed

It seemed like paths in makefile became obsolete so I tried to change the path by myself.
In OpenAvnu/examples/gstreamer-avb-plugins/gst-plugin/src/Makefile.am file we have 

libgstplugin_la_SOURCES = gstavbsink.c gstavbsink.h ../../../common/avb.c ../../../common/avb.h ../../../../lib/igb/igb.h

I tried to change some directories but I cannot find the above mentioned avb.c file anywhere. I guess the file has been renamed to some other name.

Any help is appreciated
Thanks in advance

pinealservo, 2017-11-16:
There's a reason that example is not in the top-level Makefile; as far as I can tell, the gstreamer-avb-plugin build never worked the way it was supposed to, and it's probably not worth the effort at this point of trying to build it. If you would like to use gstreamer with AVB, the avtp_pipeline code is better-integrated and actually builds.

If you would really like to put in the effort to get `gstreamer-avb-plugins` to build, the `common/avb` library was split a while back; the header `avb.h` is still usable, but it includes the individual component headers. You can find the files, from the root of the repository, at `lib/common`. The `SOURCES` line in the automake file will need to have the appropriate `avb_<component>.c` files for whichever APIs the `gstreamer-avb-plugins` code calls.

JXter, 2017-11-17:
Thank you for the reply.
Actually I tried changing that directories to lib/common and to avb_<igb,gptp,avtp>.c it did compile and build without any errors but while trying "gst-inspect-1.0 avbsink" , it showed "no element" error.
Anyway I will try with avtp_pipeline code.