Avnu/OpenAvnu #729: Corrupted frames on kernel ring error

JXter, 2017-11-28:
Hi,
I was trying to run host application with echo interface and got the error saying corrupted frames on kernel ring. I observed that some packets are received properly and then this error occurs on the listener side. Same error occured while using wav interface also. (I was able to use gstreamer interfaces successfully). I have pcap library versions on both the systems
/usr/lib/x86_64-linux-gnu/libpcap.so
/usr/lib/x86_64-linux-gnu/libpcap.so.0.8
/usr/lib/x86_64-linux-gnu/libpcap.so.1.5.3 
Every time when I'm running the program, same number of packets are received (number changes after restarting pc).
Any help is appreciated
Thanks and regards

Listener side
jx@jx-desktop:~/JXter/OpenAvnu$ sudo ./run_host.sh eth1 echo_listener_auto.ini
attaching to 0000:02:00.0
[1511852288:428218864 OPENAVB osalTime] INFO: local_time = 1511852288565855689
[1511852288:428230036 OPENAVB osalTime] INFO: ml_phoffset = 0, ls_phoffset = -214678260
[1511852288:428244731 OPENAVB osalTime] INFO: ml_freqffset = 1.000000, ls_freqoffset = 1.000029
[1511852288:428249435 OPENAVB Endpoint] INFO: AVTP Pipeline: 0.1.3 (Development)
[1511852288:428463468 OPENAVB Talker / Listener] INFO: AVTP Pipeline: 0.1.3 (Development)
[1511852288:428677518 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbMapPipeInitialize
[1511852288:428691439 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbIntfEchoInitialize
[1511852288:431168339 OPENAVB HAL Ethernet] INFO: TX buffers: 40
[1511852288:431176369 OPENAVB HAL Ethernet] INFO: IGB launch time feature is DISABLED
[1511852288:451379884 OPENAVB Endpoint SRP] INFO: detected domain Class A PRIO=3 VID=0002...
[1511852288:451389767 OPENAVB Endpoint SRP] INFO: detected domain Class B PRIO=2 VID=0002...
[1511852288:452479263 OPENAVB Listener] INFO: Attach ba:bc:1a:ba:bc:1a/11
[1511852290:678307881 OPENAVB Endpoint MAAP] INFO: MAAP address range 18 acquired:  0x91e0f000bfdc-0x91e0f000bffb (Size 32)
Send this string over the AVB network: 19202 : 0
Send this string over the AVB network: 19203 : 1
Send this string over the AVB network: 19204 : 2
Send this string over the AVB network: 19205 : 3
Send this string over the AVB network: 19206 : 4
Send this string over the AVB network: 19207 : 5
Send this string over the AVB network: 19208 : 6
Send this string over the AVB network: 19209 : 7
Send this string over the AVB network: 19210 : 8
Send this string over the AVB network: 19211 : 9
Send this string over the AVB network: 19212 : 10
Send this string over the AVB network: 19213 : 11
Send this string over the AVB network: 19214 : 12
Send this string over the AVB network: 19215 : 13
Send this string over the AVB network: 19216 : 14
Send this string over the AVB network: 19217 : 15
Send this string over the AVB network: 19218 : 16
Send this string over the AVB network: 19219 : 17
[1511852293:351176468 OPENAVB Endpoint] INFO: SRP listener callback uid=11: tlkrDecl=1
[1511852293:351326487 OPENAVB Listener] INFO: Starting stream: ba:bc:1a:ba:bc:1a/11
[1511852293:351345134 OPENAVB Raw Socket] INFO: Using *igb* implementation
[1511852293:385200074 OPENAVB Raw Socket] ERROR: pcap_next_ex failed: corrupted frame on kernel ring mac offset 0 + caplen 599977097 > frame len 336
[1511852293:385217592 OPENAVB Raw Socket] ERROR: pcap_next_ex failed: corrupted frame on kernel ring mac offset 0 + caplen 599977097 > frame len 336
[1511852293:385231057 OPENAVB Raw Socket] ERROR: pcap_next_ex failed: corrupted frame on kernel ring mac offset 0 + caplen 599977097 > frame len 336
[1511852293:385242729 OPENAVB Raw Socket] ERROR: pcap_next_ex failed: corrupted frame on kernel ring mac offset 0 + caplen 599977097 > frame len 336

Talker side
jx@jx:~/JXter/OpenAvnu$ sudo ./run_host.sh eth2 echo_talker_auto.ini
attaching to 0000:03:00.0
[1511852282:107714865 OPENAVB osalTime] INFO: local_time = 1511852284938422882
[1511852282:107723469 OPENAVB osalTime] INFO: ml_phoffset = 2900087660, ls_phoffset = -2859076988
[1511852282:107736374 OPENAVB osalTime] INFO: ml_freqffset = 0.999996, ls_freqoffset = 0.999990
[1511852282:107739893 OPENAVB Endpoint] INFO: AVTP Pipeline: 0.1.3 (Development)
[1511852282:107963457 OPENAVB Talker / Listener] INFO: AVTP Pipeline: 0.1.3 (Development)
[1511852282:108110535 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbMapPipeInitialize
[1511852282:108124843 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbIntfEchoInitialize
[1511852282:111158662 OPENAVB HAL Ethernet] INFO: TX buffers: 40
[1511852282:111170236 OPENAVB HAL Ethernet] INFO: IGB launch time feature is DISABLED
[1511852282:131447584 OPENAVB Endpoint SRP] INFO: detected domain Class A PRIO=3 VID=0002...
[1511852282:131464901 OPENAVB Endpoint SRP] INFO: detected domain Class B PRIO=2 VID=0002...
[1511852282:132466353 OPENAVB Talker] INFO: Register ba:bc:1a:ba:bc:1a/11: class: B frame size: 118  frame interval: 1
[1511852284:310628100 OPENAVB Endpoint MAAP] INFO: MAAP address range 21 acquired:  0x91e0f000a2fa-0x91e0f000a319 (Size 32)
[1511852284:335452326 OPENAVB Endpoint MAAP] INFO: Allocated MAAP address 91:e0:f0:00:a2:fa
[1511852284:335459773 OPENAVB Endpoint] INFO: SRP talker callback uid=11: lsnrDecl=fe
[1511852288:807726005 OPENAVB Endpoint] INFO: SRP talker callback uid=11: lsnrDecl=2
[1511852288:809527798 OPENAVB Talker] INFO: Starting stream: ba:bc:1a:ba:bc:1a/11
[1511852288:809539360 OPENAVB Raw Socket] INFO: Using *igb* implementation
[1511852288:833037031 OPENAVB Talker] INFO: ba:bc:1a:ba:bc:1a/11, sr-rate=4000, data-rate=4000, frames=1, size=118, batch=1, sleep=250us, sr-Kbps=4352, data-Kbps=4352
[1511852288:907767832 OPENAVB Endpoint] INFO: SRP talker callback uid=11: lsnrDecl=2
[1511852289:833123908] TX UID:11, calls=4000, frames=4000, late=0, bytes=266890, txbuf=0, mqbuf=0, 
[1511852290:833124104] TX UID:11, calls=4000, frames=4000, late=0, bytes=268000, txbuf=0, mqbuf=0, 
[1511852291:833126225] TX UID:11, calls=4000, frames=4000, late=0, bytes=270000, txbuf=0, mqbuf=0, 
[1511852292:833377387] TX UID:11, calls=4001, frames=4001, late=0, bytes=272068, txbuf=0, mqbuf=0, 
[1511852293:707196300 OPENAVB Endpoint] INFO: SRP talker callback uid=11: lsnrDecl=2
[1511852293:807213839 OPENAVB Endpoint] INFO: SRP talker callback uid=11: lsnrDecl=2
[1511852293:833379424] TX UID:11, calls=4000, frames=4000, late=0, bytes=272000, txbuf=0, mqbuf=0, 
[1511852294:833631180] TX UID:11, calls=4001, frames=4001, late=0, bytes=272068, txbuf=0, mqbuf=0, 
[1511852295:833874069] TX UID:11, calls=4001, frames=4001, late=0, bytes=272068, txbuf=0, mqbuf=0, 
[1511852296:833874281] TX UID:11, calls=4000, frames=4000, late=0, bytes=272000, txbuf=0, mqbuf=0, 
[1511852297:833874631] TX UID:11, calls=4000, frames=4000, late=0, bytes=272000, txbuf=0, mqbuf=0, 
[1511852298:834125329] TX UID:11, calls=4001, frames=4001, late=0, bytes=272068, txbuf=0, mqbuf=0, 
^C[1511852298:936946754 OPENAVB TL Host] INFO: Host shutting down
[1511852298:937136090 OPENAVB Talker] INFO: TX ba:bc:1a:ba:bc:1a/11, Totals: calls=40416, frames=40416, late=0, bytes=2737178, TXOutOfBuffs=0
[1511852300:324130686 OPENAVB Endpoint] INFO: Shutting down

JXter, 2017-12-01:
It was a bug in my libpcap version that cause error while using shorter snapshot lengths. I changed the value of the snapshot length to a larger value and now it works fine.

pinealservo, 2017-12-01:
Thanks for the update on this @JXter; I was at a loss for what might be causing it. Just so you know, there are a variety of different "raw socket" back-ends in avtp_pipeline, which have some different characteristics. You can use the ini files to select which one via the interface name; e.g. using the name `simple:eth0` or `igb:eth0`. The options are `simple`, `pcap`, `ring`, `sendmmsg`, and `igb`; some might not be available depending on architecture or compile options.

The `ring` option uses the same mechanism that libpcap does behind the scenes (AF_PACKET raw sockets transmitted/received via shared mmapped buffers between user and kernel space) so it's the most efficient option to use right now when a special driver like `igb` isn't available. It also won't do weird things to you like libpcap did.

closed 2017-12-01 by pinealservo

JXter, 2017-12-04:
Thanks for the information @pinealservo .

chuanqima, 2019-07-08:
hi,JXter
    I have a problem now, I feel that the root of your class is that my environment is directly connected to the two I210 PCs, and then the avtp_pipeline is fully compiled, but the error is reported when the command is run. I know that I compiled this and I need to manually modify the preconfigured value in the #srp option in the endpoint to 1. But after I ran it, I didn't hear any sound from listening to that end, but I saw that avb protocol messages in wireshark have been sent. Do you know this question? I look forward to your return and help, thank you.
on the talker side:
[image]
on the listener side:

JXter, 2019-07-08:
Hi,
Can't make much out of the details you shared, but on listener side you have an error in configuration. Change the value of `report seconds` to a valid value and try again.