Avnu/OpenAvnu #730: GPTP Daemon Windows 10 Testing?

gsilva2016, 2017-12-05:
Has GPTP Daemon been tested on Windows 10 with Intel 82579LM network controller? I have tried testing this with eXtreme network switch Summit x440-24t with no luck (ReturnSyncStatus always returns 0). On Win7 I do not have any issues.

christopher-s-hall, 2017-12-05:
Which driver are you using on Win10? The "inbox" driver won't work.

gabe2004, 2017-12-05:
I am testing with what Win10 installed: Provider: Intel, Driver Date: 7/25/2016, Version: 12.15.31.4, Files: e1c65x64.sys,e1c65x64.din, e1cmsg.dll, NicCo4.dll, NicInstC.dll.  I also already enabled TimeSync DWORD = 1 in the registry so I am thinking this driver may not check this registry key.

christopher-s-hall, 2017-12-05:
That sounds OK, but maybe update the driver to a newer version? 22.6 should be a recent version.

https://downloadcenter.intel.com/download/27177/Intel-Network-Adapter-Driver-for-Windows-10

I don't know of any reason older HW shouldn't work with Win10. Unfortunately, I only have newer HW for my own testing.

gabe2004, 2017-12-06:
Thanks we can close this issue. I was using the latest driver from Intel already.  However, I tested using newer hardware I219-LM and KabyLake i5 instead and worked without issues.

andrew-elder, 2017-12-06:
Great!

closed 2017-12-06 by andrew-elder

shirArg, 2018-11-29:
I am facing the same problem (I think) using I219-LM, and the latest driver from Intel site. 
I added TimeSync to the registry, but some how it seems that the driver is not reading this value- 

INFO     : GPTP [13:43:34:438] Starting IPC listener thread succeeded
Opening: rpcap://\Device\NPF_{33F88F37-D54C-4965-B2FA-EA134B078A74}
ERROR    : GPTP [13:43:34:561] Unable to determine clock rate for interface Intel(R) Ethernet Connection (2) I219-LM
ERROR    : GPTP [13:43:34:565] Failed to initialize hardware timestamper, falling back to software timestamping
STATUS   : GPTP [13:43:34:566] Starting PDelay
Error is: 6
Error is: 6
Error is: 6
Error is: 6
Error is: 6
Error is: 6
ERROR    : GPTP [13:43:34:701] Error (TX) timestamping PDelay request (Retrying-0), error=-1
Error is: 6
ERROR    : GPTP [13:43:34:704] Error (TX) timestamping PDelay request, error=-1
Error is: 6

Any idea how to solve this? 

Thanks

christopher-s-hall, 2018-11-29:
I'm working on getting this working on I219-LM. I *think* you can modify the code to use the same clock rate as I219. I'm working to verify, but in the meantime you can try modifying the code to use the same clock rate as I219.

StefanS52u1, 2021-05-06:
I'm facing the same issue. Has the problem be solved so far. Is the gPTP successfully running with the i219-LM?. If yes which modifications are required?

Ossama-MOUSTAFA, 2022-12-16:
Hi. Me too I am trying to use this with a i219-LM. 

`/**
* @brief Maps network device type to device clock rate
*/
static DeviceClockRateMapping DeviceClockRateMap[] =
{
	{ 1000000000, I217_DESC }, 
	{ 1008000000, I219_V_DESC }, 
	{ 1000000000, I210_DESC }, 
	{ 1008000000, I219_LM_DESC	}, // This one. 
	{ 0, NULL },
};`

But I just get the error "ERROR_GEN_FAILURE", when I try to get the TX hardware timestamp. (on windows)
( WindowsEtherTimestamper::HWTimestamper_txtimestamp with OID_INTEL_GET_TXSTAMP )

Does anyone have any idea about this, please ?