Avnu/OpenAvnu #740: Question regarding stream reservation

JXter, 2017-12-19:
Hi all,

I am clarifying a doubt regarding stream reservation.

If I'm using direct endpoint to endpoint connection for avb and I have multiple streams between these endpoints, does the SRP reserve bandwidth for each stream?

From what I understand, avb switches are the ones that can check availability and allocate resources, and in this case every talker advertisements will be successful even if enough resources are not available. Please correct me if I am wrong.

pinealservo, 2017-12-19:
Endpoints should monitor their own resources and not overcommit their link bandwidth with established streams. Talkers are permitted to have more Talker Advertise declarations active than available bandwidth, although I *think* they should change each to Talker Failed when available bandwidth drops below what is required for that individual stream.

Listeners on the other hand should not declare Listener Ready if establishing the stream would cause their receive bandwidth for the SR class to be over-committed. They can declare Listener Asking Failed, though, to indicate to the talker that they are interested but out of bandwidth.

This is the specified behavior, but I'm not certain it's implemented fully in the OpenAvnu software yet. Most real-world implementations have switches that will largely take care of things, or are designed so as to not violate bandwidth constraints by construction. There is a shaper daemon that works with the avtp_pipeline code that should ultimately be capable of fully managing endpoint bandwidth, but it's still in an early state.

closed 2017-12-19 by pinealservo

JXter, 2017-12-20:
Thanks for the answer. Doubt cleared.