Avnu/OpenAvnu #746: OS (and kernel) requirements

ethan-stewart-iol, 2018-01-08:
I don't see any clear requirements listed for linux distributions or kernel versions.
Is anything specific required or recommended?

I'm currently running ubuntu14.04, but may switch to something that has a newer kernel if possible.

christopher-s-hall, 2018-01-08:
Ubuntu 14.04 should be fine. That's the distribution (with updates applied) used for the automated build environment.

ethan-stewart-iol, 2018-01-10:
Ok. Are there any known issues with other other distributes/kernel versions?
Specifically, I want to use `ioctl( phc_fd, PTP_SYS_OFFSET_PRECISE, &offset )` (from daemons/gptp/linux/src/linux_hal_generic.cpp), which (I believe) requires a 4.11.x kernel, and I don't know if anything would break if I start upgrading things.

bdthomsen, 2018-01-10:
I'm not aware of any, but I haven't tried it with anything newer than a 4.4.x kernel (using Linux Mint 18.3 x64).

christopher-s-hall, 2018-01-10:
4.6 is the kernel version you need for PTP_SYS_OFFSET_PRECISE. Also, depending upon which kernel version libc is compiled against, you may need to #include include/uapi/linux/ptp_clock.h from the kernel source.

ethan-stewart-iol, 2018-01-18:
I'll move to Ubuntu 16.04 then. I don't expect any, but I'll announce any issues I run into.

closed 2018-01-18 by ethan-stewart-iol