Avnu/OpenAvnu #753: New kernel issues: updating igb-avb?

ethan-stewart-iol, 2018-01-29:
igb-avb has issues building against newer kernels. Rather than spot check them, how easy is it to sync the changes from the igb in the kernel? (which is now igb-5.4.0).

With the kernel I'm building against (4.13.0), I think there's only three issues.

pinealservo, 2018-03-01:
There are some Intel developers that periodically supply updates to keep igb-avb building with newer kernels. I'm not sure how valuable it would be to port the libigb interface to the latest upstream igb driver; it's not acceptable to the upstream kernel, and there is work in progress to provide replacements to its functionality (configuring the traffic shaping and launch time descriptors) upstream without requiring full userspace control of the queue.

Anyway, we just merged some changes to our next brach, so you might check to see if igb-avb builds for you now.

ghindman, 2018-03-01:
Yes, qav/fqtss support for mainline IGB is in kernel 4.15, and updated v3 RFC to support time-based transmission/launchtime is being prepared now for the netdev mailing list.  v2 is here: https://patchwork.ozlabs.org/cover/862639/

andrew-elder, 2018-03-01:
Great!

jfornal, 2018-03-12:
Currently the igb_avb driver already supports newer kernel versions (4.16) - compilation errors have been already resolved.

ethan-stewart-iol, 2018-03-12:
Short of actually testing anything, it builds as expected.
Marking closed.

closed 2018-03-12 by ethan-stewart-iol