Avnu/OpenAvnu #756: Problem to maintain 125us IPG with multiple talkers

chris-kuhr, 2018-02-09:
Hi *,

my problem is a variing inter packet gap in an avtp stream.
[image]

I have two avb servers back to back, each with 8 listeners and 8 talkers.
The main process has exclusive access to the igb device.
Listeners and talkers are running as threads.
The listeners use the Linux networking API.
The talkers write their avtp packets to a ring buffer, from which the main process reads them.
The uint64_t timestamps are calculated (+=125000) by the talker threads.

The timestamp are initialized in the following way:

        gPtpTimeData td;
        if(-1 == gptpinit(&igb_shm_fd, &igb_mmap)) ;
        if (-1 == gptpscaling(igb_mmap, &td));
        

Every main loop cycle I call and update the shared memory segment ptp_ts_shm:

	uint64_t now_local = 0, now_8021as=0;
	uint64_t update_8021as=0;
	unsigned delta_8021as=0, delta_local=0;
        
	if( -1 == igb_get_wallclock( igb_dev, &now_local, NULL )) ;
        
	update_8021as = td.local_time - td.ml_phoffset;
	delta_local = (unsigned)(now_local - td.local_time);
	delta_8021as = (unsigned)(td.ml_freqoffset * delta_local);
	now_8021as = update_8021as + delta_8021as;
        
	ptp_ts_shm->timestamp_nic_hw = now_local;
	ptp_ts_shm->timestamp_dma = now_local + XMIT_DELAY;
	ptp_ts_shm->timestamp_1722= now_8021as + RENDER_DELAY;

The talkers init their local timestamps at startup:

        timestamp_dma = (*ptp_ts_shm)->timestamp_dma;
        timestamp_1722 = (*ptp_ts_shm)->timestamp_1722;

During the loop, I increment the timestamps constantly. 

        timestamp_dma += 125000;
        // push packet and timestamp_dma to ringbuffer
        timestamp_1722 += 125000;

The main process writes one avtp packet per loop from each talker ringbuffer ( 1 - 8 packets per loop) to the dma memory. I have a lot of busy loops for clean dma igb_packets.
I suppose I am doing something wrong with the calculation of the correct transmission time.

Can someone help me and point me in the right direction?

BR,
Ck

andrew-elder, 2018-02-09:
I would suspect jitter in the now_local time reading. There is no guarantee (I can see) that the host CPU reads now_local exactly 125usec after the previous reading. You probably want to have something like a simulated mediaclock running.

pinealservo, 2018-02-09:
There are a number of issues here; I'm not sure exactly which is causing your observed behavior, but I don't think the system you've described will work reliably.

1. You are feeding talker streams via DMA, presumably at a rate fixed by some hardware clock. You have not accounted for the actual frequency difference between that clock and the other clocks in the system. Assuming an audio stream, it will be based on an entirely different oscillator than both the system clock and the PTP clock and thus will have some base error +/- a time-varying amount due to temperature and other factors. This has to be accounted for; first because a compliant listener will be recovering the media clock from these timestamps, and it will not work if you're lying; and second because your launch scheduling will eventually have to deal with the difference in rates.

2. The talker threads are scheduling their own frames based on when they happened to start. Scheduling in Linux does not typically have fine enough granularity or deterministic enough behavior for this to work well. Especially since in this case, the start times (which will forever determine the phase of their scheduled times, since they add a fixed amount per frame) depend on both the scheduling of the main thread (which is updating the shared memory timebase to whatever the AS wall clock read as 'now' when it happened to run) and the scheduling of the talker threads (which may start in different intervals of this sense of 'now' provided by the main thread). So the talker scheduled timestamps are going to have a fixed phase relationship determined by some random scheduling quirk at startup.

3. It's not clear from your description whether all the talkers are writing into a single ring buffer that the main thread reads from, or if there's a ring buffer per talker. I will assume it's the latter, because the former doesn't seem like it would work well. There are a number of potential difficulties here related to when the various talkers started up and how the main thread reads them out and which streaming protocol you're using. The first of these issues that comes to mind that could be causing your issues is that if you don't read them out in the correct order, then you could put out-of-order timestamps into the NIC queue, in which case the earlier-timestamped but later-queued one won't be transmitted until the earlier-queued one is. If you're reading from the talker buffers in a fixed order and queuing them to the NIC in that order, and they started at different 'now' times according to when the main thread was running, then it's very likely that some of them are out-of-order by a small amount, which would recur every interval due to the fixed phase relationship and fixed order of queueing. That could explain the delay distribution you posted.

I think the first thing to check is whether you're putting out-of-order timestamps into the NIC queue. Next, whether they're all still in the future (but not too far in the future!) according to the NIC clock when you put them in the queue. If those are both the case, you should see very even intervals, assuming you schedule the streams in the same order each time.

My suggestion would be to decouple the traffic shaping (i.e. precise launch time scheduling) from the media clock timestamping. Each talker needs to obey its SRP frames-per-interval rate, but the NIC needs to schedule frames into a single queue per traffic class, so it doesn't make sense for talkers to schedule more precisely than which observation interval a frame belongs in. If the main thread keeps track of a single start time per observation interval and queues up all the talker frames per observation interval in the same order each time, it can either time the launch times for the correct spacing or use the Qav shapers in the hardware for spacing within the interval and set them all to launch at the start of the interval.

This also allows you to use the same scheduling mechanism for both 61883-style streams and the newer AAF-style streams, which deal with media clocks that don't divide evenly into observation intervals in different ways.

Either way, you'll also need to measure and compensate for all the different clock rates in your system so that you can increment timestamp values by the correct amounts. Presentation times need to be corrected for the offset between the sample clock and the GM time, any OS sleep/wake times need to account for any difference between system time and the timebase you're using for your observation interval, and of course if your observation interval is not managed in the NIC's timebase you need to modify the launch times to account for that.

chris-kuhr, 2018-02-13:
Thank you very much for your advice!
It gave me some headache over the past days.
But now I think I have found my solution by using a dedicated mediaclock stream from an XMOS board.
It looks like a simple and elegent solution to me, in theorie - it is not ready yet.

Or am I misguided?

BR,
Ck

chris-kuhr, 2018-02-26:
Hi again,

some answers to your questions:
- The shown distribution is for one stream only.
- Yes, each talker writes to its own ringbuffer.
- AVTP packet scheduler sorts the order of those ringbuffers by the contained rx timestamps. The DMA and 1722 timestamps are calculated in the AVTP scheduler ( 125000 + talker_buf_neighbour_ts_delta ).

I have the mediaclock running and it looks much better now, but I still see the same pattern as shown above.
The peaks however, are now a lot closer to 125 microseconds - 125us and 133us.
The interesting thing is that I measured this pattern on the XMOS based mediaclock too.

I am really confused about the mediaclock, I measured it at the listener socket.
I assume from this behavior that something with the task scheduling / IRQ handling is off. 
Another assumption could be an improper configured AVB switch, which I doubt - what property should have that impact on the network?

Any further suggestions?

chris-kuhr, 2018-03-01:
Changing rtirq for igb smoothend the overall operations, but the IPG problem remains the same. 
What might cause this?
Network configuration?
Task scheduling?

andrew-elder, 2018-03-01:
Perhaps you should collect an array of something like 1000 sequential computed packet launch times and plot those and see what you see. It seems to me you need to divide this problem up if you wish to understand it better. The first question would be - am I computing the correct launch times (however that is done).

Depending on what the launch times tell you, you can then figure out the direction to take for further investigation.

chris-kuhr, 2018-03-01:
But I see this on the Mediaclock Stream too, which comes from an XMOS board. So I think it should be something different then the launch times.

andrew-elder, 2018-03-01:
Ok, so what do you think is causing the issue? Maybe there is an interfering packet on the network?

chris-kuhr, 2018-03-01:
I have no clue at the moment. Task scheduling should not interfere with the receive times of ethernet frames, or am I wrong?
My MRP allocaltion in the switch look fine ( I assume traffic shaping does as well), so how should other traffic interfere with the mediaclock stream?
I have not measured the mediaclock on a monitor switch port from a different computer yet, perhapes this might give some insight. Although I doubt it, since it would be measured on the same kind of interface/socket.

pinealservo, 2018-03-01:
There's not necessarily a problem at all, although there might be. There's no need, nor guarantee, for frames to *arrive* precisely 125usec apart. They do need to preserve the rate bounds set by their SRP reservations and maintain an average rate that corresponds to the playout rate set by the media clock timestamps so that buffers don't over- or under-run, but there's expected to be some timing jitter due to the nature of the traffic shaping that's done and how it interacts with other traffic as it passes through the network. The point is to provide bounds to the jitter rather than completely eliminate it. Newer TSN protocols can provide even tighter bounds, but at a cost of more complex traffic shaping hardware and more complex management systems.

Are you having listeners that can't correctly recover the media clock? Have you hooked talker and listener media clocks up to a scope and see if they remain locked? If that's not implemented yet, what's happening that suggests that things aren't working correctly?

If there's really a legitimate problem with what you're seeing, then maybe the following will be helpful:

How, exactly, are you capturing the IPG timestamps?  How confident are you that you're capturing the precise time of the arrival via a hardware timestamp and not a moment some time later when some system reacts to the arrival? Can you capture the timestamps directly on the other end of the talker's link rather than after the packets have been forwarded through switches?  You might try capturing at different points to see what network component is causing the variation if not the talker.  If you don't trust your timestamp capture device completely, it should be possible to hack a makeshift one that's pretty accurate out of an XMOS device; it wouldn't take a huge amount of code, but there'd be a significant learning curve figuring out where to put that code if you haven't done much development for their system before.

chris-kuhr, 2018-03-02:
I see indeed a problem. The timestamp were measured via wireshark on the listener sides. I know that this is not 100% accurate, but the relations between the measurement are. It represents the application view, sadly I do not have a scope to measure more accurately.
I should also mention, that this is a very experimental use case: trying to deploy AVB in a public internet cloud. I know that it is not the actual intention for AVB, but we want to see what happens.

The following PDFs were measured on two seperate servers connected back to back via a Extreme Networks x440t switch.
On the top the mediaclock streams, below the actual audio streams.

[image]

I find the relations very interesting, but I cannot conclude anything from it yet.
E.g. on the left side, mediaclock stream and audio stream exhibit the exact same behavior and measurement with the following statistics ( I double checked it all ):
MediaClock (left):
Sampled Timestamps: 406229
Min: 50.0000000159 usec 
Max: 199.999999978 usec
Average: 129.081692838 usec
Standard Deviation: 11.8883063331 usec

Proxy Audio Stream (left):
Sampled Timestamps: 406229
Min: 50.0000000159 usec 
Max: 199.999999978 usec
Average: 129.081692838 usec
Standard Deviation: 11.8883063331 usec

The right hand side shows the relation between mediaclock stream and audio stream of the second server. The audio stream of server two resambles the audio stream of server one pretty accurate, while the mediaclock stream does not.

MediaClock (left):
Sampled Timestamps: 706703
Min: 50.0000000017 usec
Max: 199.000000009 usec
Average: 125.857862497 usec
Standard Deviation: 18.3344940854 usec

Processing Audio Stream (left):
Sampled Timestamps: 406678
Min: 50.0000000017 usec
Max: 200 usec
Average: 131.750362695 usec
Standard Deviation: 15.9486567278  usec

The interesting thing is that all measurements show the same pattern: peaks at 125 usec and 133 usec.

Ultimately this leads to a buffer overrun, because the packet rate of 8 kHz can not be maintained. The buffer overrun is actually the problem I am investigating.
The second audio stream for example has an average of 131.750 usec IPG which results in a packet rate of 7590,13 kHz.
To maintain the IPG bounds, the average would have to be something about 125 usec. Therefore the stream would require to have the peaks somewhat below and above 125 usec (e.g. 120 usec and 130 usec). The way it is right now, there are no early packets that could compensate late packets.

pinealservo, 2018-03-02:
First of all, you should check how the Wireshark timestamps are being captured.  If the NIC on the listener supports hardware timestamping, you should be able to enable that for Wireshark. I would not try to conclude anything about direct IPG measurements without ensuring that you are measuring with hardware timestamps.  Using software timestamping could definitely cause a bimodal distribution in the IPG measurements purely due to variation in delay between hardware reception and software processing of it.

Second, there are two general things that could cause audio buffer underrun:  Either you have some instantaneous jitter in IPG that exceeds your buffer capacity to absorb, or you have a longer-term sync problem between the transmitting media clock and the receiving media clock that's completely unrelated to the IPG.  Given what you've shown in the measurements, the latter seems far more likely to be causing your problems.

The 125usec transmit interval does not actually have to be precisely synchronized with any media clock; it just has to be close enough for the SRP reservation to not be exceeded within any 125usec (for Class A traffic, at least) interval. Normally you'll be scheduling transmits based on some system or network clock, and this is probably not going to be entirely in sync with the media clock providing the audio data. If the media clock is actually a bit slower than the clock by which you're measuring IPGs, then you'd expect for the average IPG to be slightly longer than 125usec; otherwise there would not be enough data to put in the packets!

What are you doing with the media clock timestamps as they arrive to adjust the playout rate of the listener? How are you calculating the media clock timestamps on the talker? I suspect your problem is actually in one of those two places rather than in network scheduling.

chris-kuhr, 2018-04-23:
Hi again,

I did some digging.
The switch port is toggling every 10 sec between ListenerReady -> AskFailed -> ListenerReady.
This is causing the buffer overrun.
Everything else I posted above seems to be a consequence of that.
Since the streams are interrupted every 10 sec, the buffer constantly fills. Otherwise, the IPG is fine.
After some time the buffer is full and the IPG cannot be maintained anymore.

I am using the MRP query interface, I have searched my MRP Client logfiles for something out of the ordenary without any luck.
It seems to me that I have some  problem with the commands I send to the MRP daemon.

I find this hard to debug, could you give me some advice?

BR,
Ck

pinealservo, 2018-04-23:
I can sympathize, those kinds of SRP problems are really hard to debug for everyone. It might be helpful to sniff SRP traffic at multiple points if possible to see where the Asking Failed state is being introduced, and also check the software on the listener to see if it's changing its listener state itself.

I believe the Extreme Networks switches also have a serial port interface with which you can get some more SRP diagnostics that could help you track down why it is changing the reservation.

A change from Listener Ready to Listener Asking Failed across a switch typically means that there's not enough resources to make the reservation from the reservation between ports; in that case it should also be changing the matching Talker attribute to Talker Failed as it propagates towards the listener, and the Talker Failed attribute should include a failure code. Switches will also translate all registered attributes to their corresponding failed versions across a boundary port, which will occur if the port is no longer AVB Capable (problems with PTP) or the SRP Domain attributes no longer match.

Some endpoints will also register a Listener Asking Failed by themselves if they have a listener capability but have for some reason not recognized a corresponding Talker attribute yet.

Note that odd SRP bouncing behavior can occur if SRP processes are overloaded or starved of traffic somehow and can't process all the messages they need to before the protocol timeouts occur. This can be alleviated by ensuring that all your streams are numbered sequentially so that their messages can be collapsed into a single range instead of requiring an attribute per stream.

andrew-elder, 2018-04-23:
PTP not asCapable will also cause Listener Ready transition to Listener Asking Failed. As will loss of AVB domain registration. Lot of options.

chris-kuhr, 2018-04-24:
Ok...

1. Port is always asCapable (port timer is never reset)
2. MVRP is not working properly.
Switch MVRP statistics:

```
     Port  VLANs  Failed    Rx Pkt  Rx Error    Tx Pkt    Last Source    
            Regs    Regs     Count     Count     Count      Address
        2      2       0         1         0       501  a0:36:9f:bd:94:93
        3      2       0         1       268       478  a0:36:9f:bd:96:25
        9      2       0      1015         0       572  00:22:97:00:41:2c
```

Port 2 has MRP client not started yet.
Port 3 has MRP client running.
Port 9 is XMOS Dev Board

Further more, the capture on the NIC attached to Port 3 shows MVRP messages with a length of 28 bytes.
Is this correct? No padding?
Or might those be the errors the switch port is reporting?

Also, I see MVRP messages declaring VLAN 0 very often, only send from the NIC to the switch port. I cannot find anything in my code that suggests this.

Is there something about the MRP daemon I have to consider?
I am running it with systemd:
```
IOSchedulingClass=realtime
IOSchedulingPriority=1
CPUSchedulingPolicy=rr
ExecStart=/opt/avb_subsystem/daemons/mrpd/MRP_daemon -mvs -i enp4s0
```

Some more observations:

Before the state change, the bandwidth is reserved:
```
Port         Port  Effective  Class      Delta    Maximum   Reserved  Available
            Speed      Speed                    Effective                      
*3a        1000 M     1000 M  A         75.00%     75.00%      0.78%     74.21%
                              B          0.00%     75.00%      0.00%     74.21%
```

... and withdrawn afterwards:
```
Port         Port  Effective  Class      Delta    Maximum   Reserved  Available
            Speed      Speed                    Effective                      
*3         1000 M     1000 M  A         75.00%     75.00%      0.00%     75.00%
                              B          0.00%     75.00%      0.00%     75.00%
```

The  Listener Ready message of the MRP client seem to be working:

```
Port    Enabled Oper   Effective  Dplx  Jumbo  Jumbo  Cls  Bndry   State   Sr-Pvid
                           Speed                Size              App/Reg         
*3      Y       Up        1000 M  Full  N       9216  A    N      QA/IN          2
                                                      B    Y      QA/MT          2
    Talkers:
                  Stream Id           Declaration       State  
                                     Rx       Prop     App  Reg
            -----------------------  -------  -------  ---  ---
            00:22:97:00:41:2c:00:00  NA       Adv      QA   MT 

    Listeners:
                  Stream Id           Declaration       State 
                                     Rx       Prop     App  Reg
            -----------------------  -------  -------  ---  ---
            00:22:97:00:41:2c:00:00  Ready    Ready    VO   IN 
```
... but the propagation within the switch seems not to:
```
Port    Enabled Oper   Effective  Dplx  Jumbo  Jumbo  Cls  Bndry   State   Sr-Pvid
                           Speed                Size              App/Reg         
------- ------- ------ ---------  ----  -----  -----  ---  -----  -------  -------
*3      Y       Up        1000 M  Full  N       9216  A    N      QA/IN          2
                                                      B    Y      QA/MT          2
    Talkers:
                  Stream Id           Declaration       State  
                                     Rx       Prop     App  Reg
            -----------------------  -------  -------  ---  ---
            00:22:97:00:41:2c:00:00  NA       Adv      QA   MT 

    Listeners:
                  Stream Id           Declaration       State 
                                     Rx       Prop     App  Reg
            -----------------------  -------  -------  ---  ---
            00:22:97:00:41:2c:00:00  Ready    AskFail  VO   IN 
```
In both cases the talkers registrar is empty (MT), but should it not be in (IN) ?

So, all these observations point me to some errors with MVRP. But at which point do I start looking into it?
I did not see any failurecode reported in any of the captures and I'm not sure where to start poking. 
Any further advice?

BR,
Ck

chris-kuhr, 2018-05-22:
Hi,

I see MVRP messages declaring VLAN 0 very often, only send from the NICs to the switch port. 

This is one of such packets:
 ```
 17322 292.630453     IntelCor_bd:94:93     Spanning-tree-(for-bridges)_21 MRP-MVRP 60     Multiple VLAN Registration Protocol

Frame 17322: 60 bytes on wire (480 bits), 60 bytes captured (480 bits)
Ethernet II, Src: IntelCor_bd:94:93 (a0:36:9f:bd:94:93), Dst: Spanning-tree-(for-bridges)_21 (01:80:c2:00:00:21)
Multiple VLAN Registration Protocol
    Protocol Version: 0
    Message: VLAN Identifier (1)
        Attribute Type: VLAN Identifier (1)
        Attribute Length: 2
        Attribute List
            Vector Attribute
                Vector Header: 0x0001, Leave All Event: Null
                    000. .... .... .... = Leave All Event: Null (0)
                    ...0 0000 0000 0001 = Number of Values: 1
                First Value
                    VLAN ID: 0
                Attribute Event: JoinMt (3)
            Vector Attribute
                Vector Header: 0x0001, Leave All Event: Null
                    000. .... .... .... = Leave All Event: Null (0)
                    ...0 0000 0000 0001 = Number of Values: 1
                First Value
                    VLAN ID: 2
                Attribute Event: JoinIn (1)
            End Mark: 0x0000
    End Mark: 0x0000
```

This is happening inside the MRP daemon. If I use the simple_talker / simple_listener examples, everything is just working fine.
There is no obvious reason why this happens. Rebooting the machines does not help, restarting the MRP daemon neighther. But eventually it just works again ( no VLAN ID 0 in any MRP packet).

The MRP packet right before the packet above:
```
  17241 292.003171     ExtremeN_9c:52:0a     Spanning-tree-(for-bridges)_21 MRP-MVRP 60     Multiple VLAN Registration Protocol

Frame 17241: 60 bytes on wire (480 bits), 60 bytes captured (480 bits)
Ethernet II, Src: ExtremeN_9c:52:0a (00:04:96:9c:52:0a), Dst: Spanning-tree-(for-bridges)_21 (01:80:c2:00:00:21)
Multiple VLAN Registration Protocol
    Protocol Version: 0
    Message: VLAN Identifier (1)
        Attribute Type: VLAN Identifier (1)
        Attribute Length: 2
        Attribute List
            Vector Attribute
                Vector Header: 0x0001, Leave All Event: Null
                    000. .... .... .... = Leave All Event: Null (0)
                    ...0 0000 0000 0001 = Number of Values: 1
                First Value
                    VLAN ID: 2
                Attribute Event: JoinMt (3)
            End Mark: 0x0000
    End Mark: 0x0000
```

and afterwards:
```
  17353 292.892582     ExtremeN_9c:52:0a     Spanning-tree-(for-bridges)_21 MRP-MVRP 68     Multiple VLAN Registration Protocol

Frame 17353: 68 bytes on wire (544 bits), 68 bytes captured (544 bits)
Ethernet II, Src: ExtremeN_9c:52:0a (00:04:96:9c:52:0a), Dst: Spanning-tree-(for-bridges)_21 (01:80:c2:00:00:21)
802.1Q Virtual LAN, PRI: 0, CFI: 0, ID: 0
    000. .... .... .... = Priority: Best Effort (default) (0)
    ...0 .... .... .... = CFI: Canonical (0)
    .... 0000 0000 0000 = ID: 0
    Type: 802.1Q Virtual LAN (0x8100)
802.1Q Virtual LAN, PRI: 0, CFI: 0, ID: 2
    000. .... .... .... = Priority: Best Effort (default) (0)
    ...0 .... .... .... = CFI: Canonical (0)
    .... 0000 0000 0010 = ID: 2
    Type: Unknown (0x88f5)
Multiple VLAN Registration Protocol
    Protocol Version: 0
    Message: VLAN Identifier (1)
        Attribute Type: VLAN Identifier (1)
        Attribute Length: 2
        Attribute List
            Vector Attribute
                Vector Header: 0x2001, Leave All Event: Leave All
                    001. .... .... .... = Leave All Event: Leave All (1)
                    ...0 0000 0000 0001 = Number of Values: 1
                First Value
                    VLAN ID: 2
                Attribute Event: JoinMt (3)
            End Mark: 0x0000
    End Mark: 0x0000
```

Any suggestions?

BR,
Ck

pinealservo, 2018-05-23:
I guess that probably the interface between the avtp_pipeline code and the mrpd interface code is doing something odd; are you setting up just one traffic class or two?  Maybe it's registering the second one with bad values, including VID 0.

You should be able to increase logging levels and get some information about how the avb endpoint code and the mrpd process are interacting, which could give some clues as to what's really going on.

chris-kuhr, 2018-05-23:
I found a workaround.
 
First I tried just to ignore the VID 0 packet attributes in the MRP client. But the MRP daemon was not impressed.
Now When I see a VID 0  message, I send a "V--:I=0000" command to the MRP daemon and it starts behaving nicely.
The problem is not solved, since I don't know whats going on in the first place, but I found a satisfying solution to it.

closed 2018-10-02 by chris-kuhr