Avnu/OpenAvnu #757: How to use gPTP with external time reference - NTP or similar

awy, 2018-02-14:
How should gPTP be configured so that NTP, or some other external mechanism, can be used to provide synchronization with an external time reference.

christopher-s-hall, 2018-02-14:
You'll need to do some experimentation, but I would start with:
* Run phc2sys to slave the network device clock (PHC) to the system clock and discipline the system clock using NTP.
* gPTP will be able to provide the NTP derived PHC to other devices on the network.

awy, 2018-02-15:
You mean phc2sys from the linux-ptp package?
Will the network device clock (PHC) actually be available when using only software (TX) timestampting.
As far as I can see in gPTP there is no way to configure it not to discipline the system clock itself, using _adjtime_ & _adjtimex_.

pinealservo, 2018-03-01:
If you don't have hardware timestamping available, the linux-ptp package may be a better option for your synchronization tasks. Software timestamps aren't precise enough to meet the requirements of 802.1AS, which is the gPTP specification that we've focused on here.

christopher-s-hall, 2018-03-01:
"As far as I can see in gPTP there is no way to configure it not to discipline the system clock itself, using adjtime & adjtimex." The OpenAVnu daemon doesn't do this. I was suggesting using *only* phc2sys from PTP4L.

Also, my assumption was that you wanted to use gPTP to redistribute NTP/GPS time on the local network and that you had supported switches and network interfaces. If this isn't the case, @pinealservo's suggestion is the way to go.