Avnu/OpenAvnu #758: Porting gptp to Intel X550 and I350.

i-and, 2018-02-25:
Now daemons/gptp/windows/daemon_cl supports network driver for the I217-LM and I219-V adapters. Where can I get description of the driver interface for "Intel(R) Ethernet Controller X550" and "Intel(R) I350 Gigabit Network Connection"(DeviceIoControl(IOCTL_NDIS_QUERY_GLOBAL_STATS...) ) for the subsequent porting of windows/daemon_cl on X550 and I350 adapters (for Windows7/10)?

pinealservo, 2018-03-01:
Most of the Intel ethernet controller datasheets are available on the Intel site.  I haven't checked for those specific ones, but if they're not available you'd probably have to contact Intel directly.

i-and, 2018-03-05:
I have datasheets for x550 and I350. But it is not clear how to get software access to the hardware resources (time stamp, sytem timer) of the controller under Windows by analogy, as is done for the I217-LM and I219-V adapters in windows/daemon_cl:
#define OID_INTEL_GET_RXSTAMP 0xFF020264			/*!< Get RX timestamp code*/
#define OID_INTEL_GET_TXSTAMP 0xFF020263			/*!< Get TX timestamp code*/
#define OID_INTEL_GET_SYSTIM  0xFF020262			/*!< Get system time code */
#define OID_INTEL_SET_SYSTIM  0xFF020261			/*!< Set system time code */
DeviceIoControl(miniport, IOCTL_NDIS_QUERY_GLOBAL_STATS,...)

christopher-s-hall, 2018-03-05:
The windows drivers for those devices don't support those IOCTLs. Currently, support for PTP on Windows is very limited - i.e. only the PCH Ethernet controller, I217/8/9. There may be some third party solutions, but we can't help with that. You'd need to contact them directly.