Avnu/OpenAvnu #759: Aptp - master to local frequency offset is  filtered?

alecs2287, 2018-02-26:
Hello,

I am using the aptp daemon together with airplay 2 sdk  and did some testing for streaming to multiple speakers. Sometimes the ml_freqoffset reports some implausible values.
Looking at the code it seems that a filtering mechanism has been implemented:

// For now we are filtering the frequency offset until it is better tuned
		FrequencyRatio filtered;
		if (ml_freqoffset > 3.0)
		{
			GPTP_LOG_ERROR("LinuxSharedMemoryIPC::update   ml_freqoffset > 2.0: %Le", ml_freqoffset);
			filtered = fLastFreqoffset;
		}
		else
		{
			typedef FrequencyRatio FR;
			filtered = (FR(fLastFreqoffset * 63) / 64) + (ml_freqoffset / 64);
		}

		**ptimedata->ml_freqoffset = filtered;**

		ptimedata->ls_phoffset = ls_phoffset;
		**ptimedata->ml_freqoffset = ml_freqoffset;**

the ml_freqoffset is set to the filtered value but afterwards the value is overwritten to the initial value.
Is this desired behavior?

Thanks,
Alex

pinealservo, 2018-03-01:
Hi Alex,

Thanks for your interest in the aptp branch of the code!  Unfortunately, the main developers here are not very familiar with the code, as it was submitted by a third party. That's part of the reason why it's still in a separate branch; we haven't had the engineering time to do the work of thoroughly reviewing it and getting it merged in properly with the mainline of development.

What do you think about whether the code is doing the right thing? It sounds like you're skeptical, and I would be too. Does changing it seem to improve performance? If you've got some time to look into it, we'd appreciate any contributions here.

    --Levi

alecs2287, 2018-05-04:
Hi,

The issue was fixed here:
https://github.com/rroussel/OpenAvnu/commit/55457fbe98d92094e02db25739ff2719149660d0

I will close this issue.

Thanks,
Alex

closed 2018-05-04 by alecs2287