Avnu/OpenAvnu #765: Packet drops regularly every 30 seconds when running simple_talker

JohanArif, 2018-03-07:
When I run simple_talker with wireshark, I notice that the number of packets drop every 30 seconds. 

[image]

I'm assuming this could be because the pkt_sz value in simple_talker.c isn't the same as the actual packet size. What are your thoughts on this?

pinealservo, 2018-03-08:
I'm afraid there are too many possibilities to speculate.  Have you tried experimenting with different values?

JohanArif, 2018-03-19:
I tried changing the parameter **PKT_SZ** of igb_set_class_bandwidth (&igb_dev, 125000/L2_PACKET_IPG, 0, **PKT_SZ -22**, 0); from simple_talker.c
Changing PKT_SZ-22 to 98 doesn't show the drops anymore.
I changed the value to 98 based on my packet length on wireshark and based on the value of tmp_packet -> len from the link below as well
https://github.com/AVnu/OpenAvnu/blob/master/examples/simple_talker/simple_talker.c#L546

JohanArif, 2018-03-27:
Would changing the **PKT_SZ - 22** to 98 be a good idea?

kimhoemo, 2018-04-12:
@pinealservo Have you seen this problem before?

pinealservo, 2018-04-18:
I haven't used simple_talker in a very long time.  I started looking into the calculations for what the values were supposed to be here, but I ran out of time for it before finding a definitive answer as to what values should be.

What I did find:  22 is a constant that represents the size of the Ethernet headers, including the 802.1Q tag. So PKT_SZ is supposed to include those, but IIRC the SRP specification says to not include them in the tspec since it's supposed to be independent of what layer-2 you're running on.

If you change PKT_SZ such that PKT_SZ-22 == 98, does the resulting PKT_SZ make sense for the actual packet size including the Ethernet headers?

fattahzaki, 2018-07-11:
[image]

From my point of view, 22 (bytes) is the size of the MAC destination + MAC source + 802.1Q tag + Ethertype + CRC Interpacket. By that logic, PKT_SZ - 22 must be the remaining bytes for payload. Line 2103 and 2104 infer that this is true https://github.com/AVnu/OpenAvnu/blob/c253bd9204ed0fe85619266c0da1a47fbfdaab0a/lib/igb/igb.c#L2103/

Starting from line 495 in simple_talker.c https://github.com/AVnu/OpenAvnu/blob/master/examples/simple_talker/simple_talker.c#L493 we can see the construction of MAC destination, MAC source, 802.1Q tag, Ethertype, IEEE1722 and IEEE61883 packets as below:
1. Dest_addr & source_addr - 12 bytes
2. 802.1Q tag - 4 bytes
3. Ethertype - 2 bytes
4. IEEE1722 - 45 bytes
[image]
5. IEEE61883 - 57 bytes
[image]

Therefore the minimum PKT_SZ must be 12 + 4 + 2 + 45 + 57 = 120 bytes. And PKT_SZ(120) - 22 is 98 bytes. This calculation holds true for the function igb_set_class_bandwidth only. For other places where PKT_SZ is being used I'm not sure this applies.

fattahzaki, 2018-08-02:
Hi,

Any update on this? Appreciate if this issue is being accelerated as my implementation is depending on this. Thank you.

andrew-elder, 2018-08-02:
@fattahzaki - are you just testing with 2 i210s?

fattahzaki, 2018-08-02:
@andrew-elder yes