Avnu/OpenAvnu #766: travis CI build is broken

andrew-elder, 2018-03-07:


andrew-elder, 2018-03-07:
See
```

CFLAGS=-Wno-missing-braces meson lib/libavtp/ lib/libavtp/build

Meson works correctly only with python 3.5+.

You have python 3.4.3 (default, Nov 28 2017, 16:41:13) 

[GCC 4.8.4].

Please update your environment
```

Looks like a Meson issue. Anyone want to figure out how to update the version of python that travis uses?

christopher-s-hall, 2018-03-07:
There aren't really any great options. We can try to add another repository hoping nothing else breaks. This is the one I've found:

add-apt-repository --yes ppa:fkrull/deadsnakes

An example here: https://gist.github.com/instaBOT/dfb7eac79158ee8b098a90aa288d5713

Or disable Meson for now. The best option will be to upgrade to 16.04. See travis-ci/travis-ci#7260. It's not ready yet.

aguedes, 2018-03-15:
Hi guys, sorry about the delay to reply. I was on vacations :)

This issue is related to meson indeed. Meson version 0.45 was released recently and it requires Python 3.5 (see http://mesonbuild.com/Release-notes-for-0-45-0.html#python-minimum-version-is-now-35) which is not installed in Travis environment.

The best solution IMO is to stick to meson 0.44 for now. I just submitted a PR (https://github.com/AVnu/OpenAvnu/pull/767) fixing the issue.

ahogen, 2018-07-05:
@andrew-elder I think issue could be closed now.

andrew-elder, 2018-07-05:
Great, thanks

closed 2018-07-05 by andrew-elder