Avnu/OpenAvnu #770: [Question]igb_avb for arm

JXter, 2018-03-22:
Hi,

I wanted to know, whether igb_avb driver is architecture dependent and works only on x86 arch. Intel i210 is working with software shaper in arm architecture but when igb_avb is used I am getting crash reports. 

```
WARNING: CPU: 0 PID: 0 at /home/xter/Yocto_Setup/fsl-release-bsp/build-x11-imx6dlsabresd/tmp/work-shared/imx6dlsabresd/kernel-source/net/schc
NETDEV WATCHDOG: enp1s0 (igb_avb): transmit queue 0 timed out
Modules linked in: igb_avb(O) mxc_v4l2_capture ipu_bg_overlay_sdc ipu_still ipu_prp_enc ipu_csi_enc ipu_fg_overlay_sdc ov5642_camera ov5640_ca]
CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W  O    4.9.11-1.0.0+gc27010d #6
Hardware name: Freescale i.MX6 Quad/DualLite (Device Tree)
[<8010f6e4>] (unwind_backtrace) from [<8010b180>] (show_stack+0x10/0x14)
[<8010b180>] (show_stack) from [<803c0b74>] (dump_stack+0x78/0x8c)
[<803c0b74>] (dump_stack) from [<8012e020>] (__warn+0xe8/0x100)
[<8012e020>] (__warn) from [<8012e070>] (warn_slowpath_fmt+0x38/0x48)
[<8012e070>] (warn_slowpath_fmt) from [<808ac444>] (dev_watchdog+0x248/0x24c)
[<808ac444>] (dev_watchdog) from [<801821fc>] (call_timer_fn.constprop.2+0x28/0x98)
[<801821fc>] (call_timer_fn.constprop.2) from [<80182308>] (expire_timers+0x9c/0xa8)
[<80182308>] (expire_timers) from [<801823a8>] (run_timer_softirq+0x94/0x18c)
[<801823a8>] (run_timer_softirq) from [<80132a14>] (__do_softirq+0xe4/0x22c)
[<80132a14>] (__do_softirq) from [<80132e28>] (irq_exit+0xcc/0x108)
[<80132e28>] (irq_exit) from [<801710bc>] (__handle_domain_irq+0x80/0xec)
[<801710bc>] (__handle_domain_irq) from [<801014e8>] (gic_handle_irq+0x48/0x8c)
[<801014e8>] (gic_handle_irq) from [<8010bc8c>] (__irq_svc+0x6c/0xa8)
Exception stack(0x81001f30 to 0x81001f78)
1f20:                                     00000000 00000002 00000001 81000000
1f40: da23275d 0000000f ab71ceb0 00000001 181d70e6 00000010 00000004 00000050
1f60: 00000000 81001f80 80a455cc 80799200 200e0013 ffffffff
[<8010bc8c>] (__irq_svc) from [<80799200>] (cpuidle_enter_state+0x178/0x2a0)
[<80799200>] (cpuidle_enter_state) from [<80167da0>] (cpu_startup_entry+0x148/0x218)
[<80167da0>] (cpu_startup_entry) from [<80f00c58>] (start_kernel+0x37c/0x388)
```
While running gPTP on this driver, I am getting kernel message
`enp1s0 selects TX queue 3, but real number of TX queues is 1`
and the gPTP error
`ERROR    : GPTP [14:01:53:927] Error (TX) timestamping PDelay request, error=-72`
Any idea, what is causing this error?

I am thinking, the kernel message I am getting(regarding queue) is caused by some signal handling function since commenting out signal handling part didin't gave any such message. Is there is any queue related code in gPTP? Any info on this is greatly appreciated.

pinealservo, 2018-03-22:
I'm not aware of anyone else attempting to use igb_avb on non-x86 architectures, but if it compiles I don't know of fundamental reasons it shouldn't work. It's hard to say what might be going wrong from that trace. It's not technically a crash, I don't think; it's saying that the transmit queue 0 timed out, suggesting that the hardware got hung up somehow, perhaps due to bad configuration of some sort?

The other messages suggesting only one TX queue is enabled also point to bad configuration. I don't think gPTP has any way to specifically ask for different queues, so the queue mapping would be done by the default network stack mechanisms. Are you starting something else at the same time as gPTP that might be triggering the queue-related messages?

JXter, 2018-03-29:
The igb_avb driver need 4 queues for the proper working and it directs normal LAN traffic to queue 3. Therefore gPTP also try to use queue 3. Unfortunately my current configuration does not support 4 queues.

closed 2018-04-09 by JXter