Avnu/OpenAvnu #772: Compilation Error

varun-v-deshpande, 2018-03-26:
Hello, 

I am trying to compile AVB but I am getting this error :

` root@ubuntu-5151:/home/ubuntu/Downloads/Open-AVB# ./travis.sh 
make igb
mkdir -p kmod/igb && make  subdir=kmod/igb  -C kmod/igb 
make[1]: Entering directory '/home/ubuntu/Downloads/Open-AVB/kmod/igb'
make -C /lib/modules/4.13.0-37-generic/build SUBDIRS=/home/ubuntu/Downloads/Open-AVB/kmod/igb modules
make[2]: Entering directory '/usr/src/linux-headers-4.13.0-37-generic'
  CC [M]  /home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.o
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:222:13: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
   .fault  = igb_vm_fault
             ^
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:222:13: note: (near initialization for ‘igb_mmap_ops.fault’)
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:299:15: error: variable ‘igb_miscdev’ has initializer but incomplete type
 static struct miscdevice igb_miscdev = {
               ^
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:300:3: error: unknown field ‘minor’ specified in initializer
   .minor = MISC_DYNAMIC_MINOR,
   ^
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:300:12: error: ‘MISC_DYNAMIC_MINOR’ undeclared here (not in a function)
   .minor = MISC_DYNAMIC_MINOR,
            ^
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:300:12: warning: excess elements in struct initializer
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:300:12: note: (near initialization for ‘igb_miscdev’)
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:301:3: error: unknown field ‘name’ specified in initializer
   .name = "igb_avb",
   ^
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:301:11: warning: excess elements in struct initializer
   .name = "igb_avb",
           ^
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:301:11: note: (near initialization for ‘igb_miscdev’)
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:302:3: error: unknown field ‘fops’ specified in initializer
   .fops = &igb_fops,
   ^
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:302:11: warning: excess elements in struct initializer
   .fops = &igb_fops,
           ^
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:302:11: note: (near initialization for ‘igb_miscdev’)
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c: In function ‘igb_init_module’:
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:394:2: error: implicit declaration of function ‘misc_register’ [-Werror=implicit-function-declaration]
  misc_register(&igb_miscdev);
  ^
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c: In function ‘igb_exit_module’:
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:419:2: error: implicit declaration of function ‘misc_deregister’ [-Werror=implicit-function-declaration]
  misc_deregister(&igb_miscdev);
  ^
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c: In function ‘igb_set_interrupt_capability’:
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:1087:10: error: implicit declaration of function ‘pci_enable_msix’ [-Werror=implicit-function-declaration]
    err = pci_enable_msix(pdev,
          ^
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c: At top level:
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:2351:21: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .ndo_set_vf_vlan = igb_ndo_set_vf_vlan,
                     ^
/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.c:2351:21: note: (near initialization for ‘igb_netdev_ops.ndo_set_vf_vlan’)
cc1: some warnings being treated as errors
scripts/Makefile.build:308: recipe for target '/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.o' failed
make[3]: *** [/home/ubuntu/Downloads/Open-AVB/kmod/igb/igb_main.o] Error 1
Makefile:1550: recipe for target '_module_/home/ubuntu/Downloads/Open-AVB/kmod/igb' failed
make[2]: *** [_module_/home/ubuntu/Downloads/Open-AVB/kmod/igb] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-4.13.0-37-generic'
Makefile:262: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Leaving directory '/home/ubuntu/Downloads/Open-AVB/kmod/igb'
Makefile:36: recipe for target 'igb' failed
make: *** [igb] Error 2
root@ubuntu-5151:/home/ubuntu/Downloads/Open-AVB# 
`
The details of my system are : ` Linux ubuntu-5151 4.13.0-37-generic #42~16.04.1-Ubuntu SMP Wed Mar 7 16:03:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux `

How can I solve it ? Is this kernel supported by AVB ?

andrew-elder, 2018-03-26:
What branch are you compiling?

varun-v-deshpande, 2018-03-26:
Master Branch.

andrew-elder, 2018-03-26:
Try open-avb-next branch

varun-v-deshpande, 2018-03-28:
Yes, it works, Thank you
(Sorry to inform you late)

closed 2018-03-28 by andrew-elder