Avnu/OpenAvnu #774: [Question] Using gPTP for accurate synchronization over network bridge?

martomi, 2018-03-31:
Hi everyone,

I have 2 ethernet cards (Intel i210) and have created a bridge between them in Ubuntu 16.04.
Connected to those ethernet cards I have 2 devices which need to synchronize over PTP.
One of the connected devices is master, the other one - slave.

Without any additional setup, the accuracy of synchronization I achieve is only 100-300 microseconds. I'm aiming to achieve accuracy below 1 microsecond.

I've been reading about gPTP and AVB, but couldn't quite figure out if the tools in this project can be used for this use case? Has anyone attempted doing this? I would appreciate any hints on the setup. 

I've followed the instruction on setting up the igb-avb driver for i210 and tried running the gptp daemon but had no success. (I was getting time out messages whenever I try to run it.)

Do I actually need to run the same gptp protocol on the other devices as well or just on the bridge? The other devices in my setup implement the standard IEEE 1588.

Thanks a lot!

martomi, 2018-04-02:
I was looking into potentially using the i210's SDPs to sync them.

> The I210 has four software-defined pins (SDP pins) that can be used for IEEE1588 auxiliary device connections, enable/disable of the device, and for other miscellaneous hardware or software-control purposes. These pins can be individually configurable to act as either standard inputs, General-Purpose Interrupt (GPI) inputs or output pins.

(Excerpt from: [i210's Datasheet](https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/i210-ethernet-controller-datasheet.pdf))

However I was hoping that the controller's support for 802.1AS and OpenAvnu can provide higher level abstraction to accomplish that.

andrew-elder, 2018-04-02:
Yes, I agree, those SDPs would seem like the best way to sync the two clocks. I'm not aware of software support to actually do this - but may be there is....

@christopher-s-hall - any comments on the state of software support for using SDP pins in Linux?

christopher-s-hall, 2018-04-02:
The I210 SDP pins are well supported in Linux at the PTP driver. Take a look at PTP_PIN_SETFUNC in linux/ptp_clock.h. You'd need to write another software application to make the adjustments syncing the clocks. It's possible that phc2sys (part of PTP4L) might sync the clocks also, but it would not work as well as using the SDP pins.

martomi, 2018-04-03:
Thanks a lot for the responses, I've looked more into phc2sys. Seems like it works well enough (<150ns error) for my current needs, so I'll be using it to test the setup for now.

Let's assume I have the clocks of the two i210's synced with phc2sys. How do I proceed to configure PTP such that the delays introduced in switching inside the bridge are accounted for? 

Would the gptp daemon from this project be suitable for this task? I've tried using ptp4l without success, I think it doesn't handle this kind of bridge setup.

```EXCEPTION: GPTP [...] Delay Response Receipt Timeout```

is the message I get from gptp.

martomi, 2018-04-18:
Just in case anyone ever stumbles upon this thread: Software bridging on linux doesn't seem to be compatible with any of the PTP implementations. I ended up going different route that doesn't involve a bridge.

closed 2018-04-18 by martomi

jdkoftinoff, 2018-04-18:
FYI: 802.1AS is transported as a bridging protocol itself. This means that bridges are not supposed to forward 802.1AS ethernet frames, as per 802.1D.  The 802.1AS protocol is a peer to peer protocol; A bridge needs to be aware of 802.1A to work with it. i.e. the Bridge itself needs to run the gPTP protocol state machines to participate. With this in mind, the software bridging implementation on linux is compliant, and is doing the correct thing.