Avnu/OpenAvnu #775: [Question] Sample applications to upstream

fattahzaki, 2018-04-02:
Hi,

I have two sample applications I'm planning to upstream. Below is the high level description of the sample apps:

1. The first one is to demonstrate the gPTP time synchronization accuracy between devices. It'll initialize and output PPS on SDP0 of i210 and read timestamps on SDP1. It'll also setup OPC UA server on every device to communicate its timestamp data to the Grand Master which is the OPC UA client. Timestamp data will be aggregated and recorded in a log file in Grand Master where it'll be used to plot a graph (via gnuplot) that highlights the slave's timestamps drift from Grand Master's timestamps.

2. The second sample app is to demonstrate the i210 LaunchTime and 802.1Qbv switch gating mechanism (see figure below). The 802.1Qbv switch will "open" its gate in a scheduled window where only TSN frames can pass through and other frames are blocked ensuring better time efficiency. The sample app sends and receives TSN and best effort frames while the traffic generator creates congestion in the network. A file will be generated to dump the latency data of the frames on each endpoint. The file is then being used to plot a graph (via gnuplot) that compares the latency of TSN and best effort frames as well as the packet loss information on each endpoint.

[image]

Do you think it's okay to place these sample applications inside /examples folder? Let me know what you think, or if you need any additional information. 

Thanks.

andrew-elder, 2018-04-02:
Yes, examples folder would be fine. Thank you for contributing. Maybe add a readme.md file to that dir as well.

fattahzaki, 2018-04-03:
Noted. Open-avb-next should be the branch to upstream to, no?

pinealservo, 2018-04-03:
Yes, all new work goes into open-avb-next.

kalkhour, 2018-07-03:
Hi,
I was wondering if you have already uploaded these two applications. I can't find them in the examples folder of open-avb-next.
Thanks