Avnu/OpenAvnu #776: Possible dead lock when handling SYNC_RECEIPT_TIMEOUT_EXPIRES event

freddeng, 2018-04-16:
Hi,

I am using the PTP daemon (from https://github.com/rroussel/OpenAvnu) with AirPlay2 SDK and found a possible dead lock when SYNC_RECEIPT_TIMEOUT_EXPIRES is received.

Although this was found in the branch (https://github.com/rroussel/OpenAvnu), I assume the Master branch will be the same as this part of the code is the same.

What happens is that 
1. CommonPort::stopSyncReceiptTimer() or CommonPort::startSyncReceiptTimer() gets hold of syncReceiptTimerLock before calling clock->deleteEventTimerLocked()/addEventTimerLocked();
2. then SYNC_RECEIPT_TIMEOUT_EXPIRES is received and handled in LinuxTimerQueueHandler() and that gets hold of timerq->lock, later in the callback CommonPort::processSyncAnnounceTimeout() it will try to call startSyncReceiptTimer() which be blocked by step 1, in the meantime step 1 will be also blocked by the timer callback because it waits for timerq->lock.

I think normally startSyncReceiptTimer/stopSyncReceiptTimer will finish in a very short time so maybe it's not very obvious. But in my environment it looks like startSyncReceiptTimer/stopSyncReceiptTimer sometimes take ~100 ms to execute "syncReceiptTimerLock->lock()"so this happens quite often.

In my opinion, we could fix this by using just one lock, the one in TimerQueue, like add a new method in IEEE1588Clock called "readdEventTimerLocked":
void IEEE1588Clock::readdEventTimerLocked
(CommonPort *target, Event e, unsigned long long time_ns)
{
    if( getTimerQLock() == oslock_fail ) return;
	deleteEventTimer( target, e);
	addEventTimer( target, e, time_ns );
    if( putTimerQLock() == oslock_fail ) return;
}
And remove syncReceiptTimerLock completely from CommonPort to avoid this dead lock but I do not know if that is going to cause other side effects. 
Also maybe similar dead lock could happen for CommonPort::startSyncIntervalTimer and CommonPort::startAnnounceIntervalTimer but I am not entirely sure.

christopher-s-hall, 2018-04-16:
I'll have to take a closer look at the code, but I suspect syncReceiptTimerLock and friends were added when we added processing for signalling messages that modify the timers outside of the handler. I'm not immediately sure of the implications of removing those locks.

Would something like the following work:

void CommonPort::startSyncReceiptTimer
( long long unsigned int waitTime )
{
        clock->getTimerQLock()
        syncReceiptTimerLock->lock();
        clock->deleteEventTimer( this, SYNC_RECEIPT_TIMEOUT_EXPIRES );
        clock->addEventTimer
                ( this, SYNC_RECEIPT_TIMEOUT_EXPIRES, waitTime );
        syncReceiptTimerLock->unlock();
        clock->putTimerQLock()
}

freddeng, 2018-04-17:
Yes, that also works and is probably a safer option.
Thank you.

andrew-elder, 2018-04-17:
@freddeng , do you have time to implement @christopher-s-hall's suggestion, test it, and create a pull request to opan-avb-next branch?