Avnu/OpenAvnu #784: Does this repo Supports TSN protocols ?

Ravi-Teja-konda, 2018-05-18:
Hello,

I have a question regarding TSN protocol support for AVB. I'm hoping someone can help me with:

we have developed AVB on Linux and trying to update the AVB to TSN by adding TSN protocols.
when I went through the directories of repo I've not seen the TSN related Protocol directories.
Does this repo support TSN or only AVB?

I have just entered the world of TSN. Any help will be highly appreciated. Thanks.

Regards,
Ravi Teja.

Ravi-Teja-konda, 2018-05-21:
Hello Levi Pearson,

Thank you for your reply. After reading the Open-AVB Status page  [https://github.com/AVnu/OpenAvnu/wiki/Open-AVB-Status], I came to know that this repo supports  IEEE802.1AS-2011. which is a part of TSN. 

Is it possible to implement all the remaining 802.1Qbu, 802.1Qbv, 802.1Qcc, 802.1CB, 802.1Qch, 802.1Qci standards with this existing repo?

Are these standards still under development?

if possible can you please send me the link to these standards...

pinealservo, 2018-05-22:
I don't think the amount of work required fits the definition of "possible" that you're looking for, although some would be easier to build from what's here than others would. We would certainly be happy to have people help move the SRP implementation towards the newer TSN functionality or contribute implementations of other protocols, but there is not currently a lot of active development going on here.

I'm not sure what the current status of those standards are, but you can look it up on the IEEE website. If you are interested enough, there is even enough information on the IEEE website to show you how to participate in the ongoing development of the protocol standards, which would allow you to have access to draft versions. I can't provide direct links to any of them, though.

pinealservo, 2018-05-22:
You can see the state of TSN standard updates here: https://en.wikipedia.org/wiki/Time-Sensitive_Networking

There are extensions to SRP in TSN that are required for configuring some of the new TSN functionality and which provide different methods of doing overall system configuration. I don't know that you'd have to start from scratch, but you would have to implement the new SRP functionality.

Ravi-Teja-konda, 2018-05-22:
Thank you for the quick response.

chris-kuhr, 2018-05-22:
Hi *,

I could be of some assistance for the TSN implementation.
Not right away, but a year from now.
At the Moment our research project deals with AVB exclusively. In the future we might also want to use TSN in our research project.

BR,
Ck

anishkhannayellapu124, 2024-06-10:
Team,
Thank you for this base. Does this support TSN(Qav,Qbv)
. how to use traffic control, when we expose queues to user space.