Avnu/OpenAvnu #785: normal traffic consuming more bandwidth

JXter, 2018-05-18:
I am running openavb_harness in two PC's with igb alon with gPTP and MRP daemons. I am able to run with all the interfaces but when I test the bandwidth consumption I am getting confused.

I am checking the bandwidth consumption with nload.
When running mjpeg_gst_talker/listener the bandwidth is showing 10Mbits/s(Max) which is fine, then I add some junk packets with a packet generator and try to send at maximum rate. Now the bandwidth usage goes to 943Mbits/s . 

While running avb, it should be in Qav mode and only 25% should be available to normal traffic, yet its taking almost 94%.  Is there any better way to verify the working of credit based shaping functionality of avb? 

Talker log
```
sudo ./openavb_harness -I eth1 mjpeg_gst_talker.ini
attaching to 0000:02:00.0
[1526627373:352248072 OPENAVB osalTime] INFO: local_time = 1526627374746203714
[1526627373:352256549 OPENAVB osalTime] INFO: ml_phoffset = 0, ls_phoffset = -1483751214
[1526627373:352264198 OPENAVB osalTime] INFO: ml_freqffset = 1.000000, ls_freqoffset = 1.000028
[1526627373:352267130 OPENAVB Endpoint] INFO: AVTP Pipeline: 0.1.3 (Development)
[1526627373:376683982 OPENAVB Talker / Listener] INFO: AVTP Pipeline: 0.1.3 (Development)
[1526627373:379943485 OPENAVB HAL Ethernet] INFO: TX buffers: 40
[1526627373:379950829 OPENAVB HAL Ethernet] INFO: IGB launch time feature is DISABLED
Opening: mjpeg_gst_talker.ini,ifname=eth1
Configuring: mjpeg_gst_talker.ini,ifname=eth1
Starting: mjpeg_gst_talker.ini,ifname=eth1
[1526627373:629591243 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbMapMjpegInitialize
[1526627373:629615421 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbIntfMjpegGstInitialize
[1526627373:629798635 OPENAVB Talker] INFO: Register ba:bc:1a:ba:bc:1a/5: class: A frame size: 1436  frame interval: 1
[1526627374:822112599 OPENAVB Endpoint] INFO: SRP talker callback uid=5: lsnrDecl=2
[1526627374:826533401 OPENAVB Talker] INFO: Starting stream: ba:bc:1a:ba:bc:1a/5
[1526627375:207541197 OPENAVB MJPEG Interface] INFO: Pipeline: v4l2src ! video/x-raw,width=640,height=480 ! jpegenc ! rtpjpegpay ssrc=5 timestamp-offset=1 seqnum-offset=1 ! appsink name=avbsink
[1526627375:293038126 OPENAVB Raw Socket] INFO: Using *igb* implementation
[1526627375:350240193 OPENAVB Talker] INFO: ba:bc:1a:ba:bc:1a/5, sr-rate=8000, data-rate=8000, frames=1, size=1436, batch=1, sleep=125us, sr-Kbps=93056, data-Kbps=93056
[1526627381:701815118 OPENAVB Endpoint] INFO: SRP talker callback uid=5: lsnrDecl=2
[1526627381:801490169 OPENAVB Endpoint] INFO: SRP talker callback uid=5: lsnrDecl=2
[1526627384:837223532 OPENAVB Endpoint] INFO: SRP talker callback uid=5: lsnrDecl=2
[1526627385:038601787 OPENAVB Endpoint] INFO: SRP talker callback uid=5: lsnrDecl=2
[1526627385:138382296 OPENAVB Endpoint] INFO: SRP talker callback uid=5: lsnrDecl=2
[1526627395:228810185 OPENAVB Endpoint] INFO: SRP talker callback uid=5: lsnrDecl=2
[1526627406:982399011 OPENAVB Endpoint] INFO: SRP talker callback uid=5: lsnrDecl=2
[1526627417:724963649 OPENAVB Endpoint] INFO: SRP talker callback uid=5: lsnrDecl=2
[1526627430:745459381 OPENAVB Endpoint] INFO: SRP talker callback uid=5: lsnrDecl=2
[1526627442:238739389 OPENAVB Endpoint] INFO: SRP talker callback uid=5: lsnrDecl=2
```
Thank you

pinealservo, 2018-05-19:
The credit-based shaper only applies to SR-class traffic, i.e. your reserved streams. The amount of bandwidth you're allowed to reserve is capped at 75%, which means that if you do reserve all of it and actually use the reservation, then non-SR traffic will be limited to 25% of the bandwidth. But generally, non-SR traffic can continue to use whatever bandwidth is leftover after the SR-traffic goes; there's absolutely no limit placed on it by AVB.

JXter, 2018-05-21:
That helps a lot. I have one more query regarding the cbs functionality. I believe the flow of packet while using igb rawsock and ring rawsock(and other rawsockets too) varies so if I use ring rawsocket implimentation will I get the desired traffic shaping? In ring rawsocket implimentation I havent seen any code to take care of the queue choice.

pinealservo, 2018-05-22:
The rawsock code doesn't touch the shaper hardware at all; it does its best to deliver packets to the NIC as efficiently as possible, but it may be configured to deliver them in chunks and rely on the kernel or NIC to do further shaping. Assignment to particular queues happens elsewhere as well.

There should be some igb-specific code for configuring the cbs shaper registers in the i210 NIC somewhere in the avb-pipeline code, but when the igb library isn't compiled in there's currently only the Linux kernel's native queuing and shaping functionality to rely on, and right now there's just a preliminary implementation of how that could be done in the shaper daemon.

JXter, 2018-05-22:
In igb rawsocket I have seen a queue index parameter passed with igb_xmit function, this parameter is used as index for some buffer rings. This is why I thought igb rawsocket is important for cbs.  
How can I check whether cbs shaping is happening or not?

pinealservo, 2018-05-22:
Getting the data into the right queue would be important for shaping, yes.

You can tell if cbs is working if your SR-class frames remain evenly spaced on the wire even when you enequeue them in chunks, and also when that rate remains fairly consistent in the presence of a lot of best-effort traffic trying to share the interface.

You can also see about checking the current i210 configuration register values to see whether it has turned on the shaper. I can't remember if there's a way to dump that via ethtool, but you could always add some debugging log statements to the igb driver to see when its shaping configuration changes.

JXter, 2018-05-28:
[image]

This is the packets per second that I am getting.
As you can see the avb traffic is going down when I am send some other packets. Here I am using alsa talker. Its Class A traffic, so without any interfering traffic its constant at 8000 packets/sec. I have captured this with wireshark.

I have checked the value that is written to TQAVCTRL and its correct. I suspect its some bad configuration issue.
Can you please tell me, probably where its going wrong?

JXter, 2018-05-30:
Thanks @pinealservo .I will look into the problem, with the info you shared.

Ravi-Teja-konda, 2018-07-19:
Hello JXter,

I need your help in running the mjpeg_gst_talker , listener .I ran mpeg2ts codec streaming which is working fine. but when i run mpeg codec streaming i'm losing packets at listener side. i read your log of talker which is running igb implementation. Is it mandatory to run mjpeg with igb implementation . because i'm using pcap implementation to run mjpeg and i have no idea of IGB launch time ..what is  IGB launch time feature ?

Thans in advance,
Ravi Teja.

JXter, 2018-07-23:
Hi Ravi-Teja-konda,
Just for running openavb_harness with mjpeg interfaces doesn't need igb, pcap should do just fine. IGB launch time is for deciding at what time a packet should be sent from the NIC.
Since you tried mpeg2ts interfaces without any lost packets, I think the error might be in your mjpeg configuration files(mjpeg ini files). Check for any mismatching parameter values. Thats all I can say, based on what you said.