Avnu/OpenAvnu #786: Listener/Talker example Raspberry pi2

TengXu94, 2018-05-21:
Dear all,
I'm following this project since a while and I would like to know if there are any chance to run the simple listener/talker example described in /documents/examples using two raspberries. 
I also know that in order to run the gptp technology we need a required ethernet adapter in the specific I have found on the issues the Intel l210, but I would like to know if there are is a way to bypass this problem on the raspberries, with lower performance obviously (the time synchronization is not guaranteed in a small microseconds window (?)).

Thank you for you attention
Best regards

andrew-elder, 2018-05-21:
I don't think this is possible. You could do something to transfer audio, but it wouldn't be AVB. 802.1AS states that a port is asCapable when the measured pDelay is 800 nanoseconds or less. Without hardware time-stamping of PTP packets (which the Raspberry Pi does not have), there isn't a good way to compute the pDelay with sub microsecond accuracy. The Beagle Bone family does support h/w time stamping and so do any iMX6/iMx7 based deices. They would be a better choice for trying things out.

TengXu94, 2018-05-21:
Got it,
thanks for the fast reply andrew-elder

closed 2018-05-23 by TengXu94