Avnu/OpenAvnu #787: Issues with Video Stream via Avtp Pipeline

Markus-E-Gruber, 2018-05-30:
Hello everyone,

i have two Endpoints connected directly without a switch and i managed to build this repository correctly and can run Simple_talker/Simple_Listener and the Avtp Pipeline.
I am using "open-avb-next" branch.
My Goal is to stream Video and Audio via the avtp Pipeline and to play the incoming AV Stream immediately at the Listener Endpoint.
I am able to stream audio via the AVTP Pipeline with "openavb_harness" on Talker with the "wav_file_talker.ini" and on the Listener side "openavb_harness" with "alsa_listener.ini". This is the "61883-6" example in the Avtp Pipeline README file. This works well but the sound quality is really bad. Maybe i have to change some parameters.
```

# 61883-6 talker
sudo ./openavb_harness -I $IFNAME -s $STREAMS -d 0 -a a0:36:9f:2d:01:ad wav_file_talker.ini,sr_class=$CLASS,map_nv_tx_rate=$RATE,max_transit_usec=$TRANSIT_USEC,intf_nv_file_name=test01.wav,report_seconds=$REPORT
# 61883-6 listener
sudo ./openavb_harness -I $IFNAME -s $STREAMS -d 0 -a a0:36:9f:2d:01:ad alsa_listener.ini,sr_class=$CLASS,map_nv_tx_rate=$RATE,max_transit_usec=$TRANSIT_USEC,report_seconds=$REPORT
```

But my main problem is streaming a video file and especially playing the video immediately at the Listener while the Talker sends the videostream.

I think the following Example should do this, but i think the "mpeg2ts_gst_listener.ini" at the Listener side is not working and it is not even in this directory anymore. 
I found it at OpenAvnu/lib/avtp_pipeline/platform/Linux/intf_mpeg2ts_gst/mpeg2ts_gst_listener.ini. When i use it i get the following Error:

```
~/TSN/OpenAvnu/lib/avtp_pipeline/build/bin$ sudo ./openavb_harness -I enp1s0 -s 1 -d 0 -a a0:36:9f:2d:01:a mpeg2ts_gst_listener.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,report_seconds=0
Opening: mpeg2ts_gst_listener.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,report_seconds=0,dest_addr=91:e0:f0:00:fe:00,stream_addr=a0:36:9f:2d:01:a,stream_uid=0,ifname=enp1s0
Configuring: mpeg2ts_gst_listener.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,report_seconds=0,dest_addr=91:e0:f0:00:fe:00,stream_addr=a0:36:9f:2d:01:a,stream_uid=0,ifname=enp1s0
Error configuring: mpeg2ts_gst_listener.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,report_seconds=0,dest_addr=91:e0:f0:00:fe:00,stream_addr=a0:36:9f:2d:01:a,stream_uid=0,ifname=enp1s0
[1527605859:030316559 OPENAVB osalTime] INFO: local_time = 1527605858111873574
[1527605859:030342231 OPENAVB osalTime] INFO: ml_phoffset = -3204526960, ls_phoffset = 827287291
[1527605859:030354997 OPENAVB osalTime] INFO: ml_freqffset = 1.000004, ls_freqoffset = 1.000000
[1527605859:030362318 OPENAVB Talker / Listener] INFO: AVTP Pipeline: 0.1.3 (Development)
[1527605859:030685597 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbMapMpeg2tsInitialize
[1527605859:030703465 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbIntfMpeg2tsGstInitialize
[1527605859:030723326 OPENAVB Talker / Listener] **ERROR: Interface initialize function lookup error: ./openavb_harness: undefined symbol: openavbIntfMpeg2tsGstInitialize.**
[1527605859:030728589 OPENAVB Talker / Listener] **ERROR: Failed to open mapping / interface library**
```

Is there any Problem with the gstreamer implementation? Is it possible to implement the gstreamer successfully? In other issues it was said that the gstreamer plugin is an abandoned project and never really worked properly #725.

Can you help me to stream and play a Videostream properly via the avtp Pipeline?

Thank you :)

PS: I can stream a .ts video and store it in a .ts file at the Listener by using the "mpeg2ts_file_listener.ini", but i would like to play it immediately.

pinealservo, 2018-05-30:
The gstreamer plugin in the examples directory mentioned in that issue is an entirely different one from the one in the avtp_pipeline code. I'm sure the pipeline one has worked before on someone's system, but I haven't done anything with it myself.

The error you are seeing is from the avtp_pipeline plugin resolution process, which takes a symbol name from the ini file and tries to look up its address in the process at runtime; in this case, it can't find it. That could either be because it wasn't built into the binary, or because there's a typo and the name in the file doesn't match the actual symbol name.

You'll want to first check to see if the gstreamer plugin got built, and then check to see whether you've actually got the correct function name in the ini file.

Markus-E-Gruber, 2018-05-30:
Thank you for the quick response!

So there is another gstreamer plugin in the avtp pipeline?
I think some dependencies to the gstreamer plugin were not built into binary as you mentioned!
I built the avtp pipeline like in the description in the README file and without SRP Support with $ AVB_FEATURE_ENDPOINT=0 make avtp_pipeline.
Should the gstreamer plugin be built automatically while building the avtp pipeline? Or are there some additional building procedures necessary to get this gstreamer plugin working?
I can't see any hints on building a gstreamer plugin in the avtp pipeline directory.

Regards

athul7pk, 2018-05-30:
in the latest changes by default GSTREAMER feature is turned off , you can enable it by simply modifying 
https://github.com/AVnu/OpenAvnu/blob/master/lib/avtp_pipeline/**avtp_pipeline.mk**

change AVB_FEATURE_GSTREAMER ?= 0
 to **AVB_FEATURE_GSTREAMER ?= 1;**
rebuild it with

**make clean;AVB_FEATURE_ENDPOINT=0 make avtp_pipeline ;**

and lemme know if you have any problem ?
you have to modify the talker/listner.ini file accordingly to take the shared libraries
please use the appropriate .ini file mpeg2ts_gst_listener.ini

Ravi-Teja-konda, 2018-05-31:
Hello Everyone,

I'm facing the the same issue when i run the video listener with mpeg2ts_gst_listener.ini i built the avtp_pipline by making  AVB_FEATURE_GSTREAMER ?= 1; in avtp_pipeline.mk and rebuilt it  by  
make avtp_pipeline after running i see no ouput .. 

i am running run_avtp_pipline.sh,run_maap.sh,run_igb.sh,run_srp.sh daemons  and running listener using below command  
[~/OpenAvnu/OpenAvnu-master/lib/avtp_pipeline/build/bin$ sudo ./openavb_harness -I enp3s0 alsa_listener.ini,stream_addr=ba:bc:1a:ba:bc:1a] same as echo_listener.sh example does this needs to be changed because i have tested audio with same command it was working fine

any thing need to be changed please suggest...

Thanks, In advance.

Ravi-Teja-konda, 2018-05-31:
Hello Athul,
Thank you for your reply. i started running the video test as above mentioned same issue is getting generated i.e., NO AVTP packets in the wireshark..

I am sharing logs and .ini files 
[LOG.tar.gz](https://github.com/AVnu/OpenAvnu/files/2058594/LOG.tar.gz) 
Please suggest if any thing goes wrong ...

Waiting for your reply...

athul7pk, 2018-06-01:
for initial verification rebuild with 
make clean;
AVB_FEATURE_ENDPOINT=0 make avtp_pipeline;

please modify your configuration file(*.ini) with ifname = enp3s0 (Assume you have an interface enp3s0  with igb_avb loaded (you can confirm with lspci -v )) 
Then rerun application it will work ,If not pls share the logs in comment itself

Ravi-Teja-konda, 2018-06-01:
Hello Atul,

Thank you very much for your reply.After making AVB_FEATURE_GSTREAMER ?= 1; and rebuilding and modifying  the .ini  files 
I made  mpeg2ts_file_listener.ini and mpeg2ts_file_talker.ini work .
But when run mpeg2ts_gst_listener.ini on the listener side i get this error

[[1527859600:841440892 OPENAVB AVB Stack] ERROR: Error creating pipeline: no element "mpegtsdemux"]
 I can see the AVTP packets in the wireshark but unable to get the output on the application..
Is this error related to gstreamer or application?

does this  mpeg2ts_gst_listener.ini  produce output in live or just write in to file ?

by the way do i need to do the below mentioned thing ? for testing 
================================================
for initial verification rebuild with
make clean;
AVB_FEATURE_ENDPOINT=0 make avtp_pipeline;
==================================================

Waiting for your reply..

athul7pk, 2018-06-01:
AVB_FEATURE_ENDPOINT=0 for disabling mrp and srp  it will avoid complexity for troubeshooting.

 MPEG transport stream parser is missing in your gstreamer plugins install that 
plugins providing mpegtsdemux .
it is part of bard plugin if my memory is correct 

sudo apt-get install gstreamer0.10-plugins-bad ..

Ravi-Teja-konda, 2018-06-01:
Thank you Atul for quick reply.
I will try it and inform if any issues...

Ravi-Teja-konda, 2018-06-04:
Hello Atul,

I've tested the mpeg2ts_gst_listener and mpeg2ts_gst_file_talker.ini by upgrading listener to gstreamer 1.14.1 and the error got resolved i have tested it by uncommenting [intf_nv_gst_pipeline = appsrc name=avbsrc ! filesink location=/dev/shm/listen.ts] it getting stored..

However, i need to test the gst_listener by opening up the window automatically when there is a AVTP data on the listener side 

Can i do it through gstreamer commands? or is it supposed to be stored only in a file and read?

Waiting for your reply..

Ravi-Teja-konda, 2018-06-05:
Hello everyone,

Can some one please let me know what to do to get the incoming video playing automatically on the Listener ?
Any comments would be of great help.

Thank you!

Ravi-Teja-konda, 2018-06-14:
Hello Markus,

Did your issue got resolved ?

Any comments would be of great help.

Thank you!

Markus-E-Gruber, 2018-06-15:
Hi @Ravi-Teja-konda ,
sorry the gstreamer implementation does not work for me yet.

I'm not able to to play the incoming IEEE1722 video stream with "mpeg2ts_gst_listener.ini" on the Listener. I can see the 1722 packets in wireshark on the Listener but i can't see the video...

I also tried the "h264_gst_talker.ini" and "h264_gst_listener.ini" but without any success.

The ALSA implementation works fine for me:
```
Talker:
sudo ./openavb_harness -I enp2s0 wav_file_talker.ini

Listener:
sudo ./openavb_harness -I enp1s0 alsa_listener.ini 
```
On the Listener i can hear the wav sample through the headphones/speaker.

Storing a Video in a .ts file works also fine:
```
Talker:
sudo ./openavb_harness -I enp2s0 mpeg2ts_file_talker.ini

Listener:
sudo ./openavb_harness -I enp1s0 mpeg2ts_file_listener.ini
```

I would appreciate any hints to get the Video playing automatically on the Listener while the Videostream is ongoing.

Thank you!
Best Regards

Ravi-Teja-konda, 2018-06-18:
Hi @Markus-E-Gruber 
I am trying to check OpenAvnu by compiling it with AVB_FEATURE_ENDPOINT=0 however i am able to run it on the talker side but I'm getting Aborted(core dumped) on the listener. Do you have any info related to command explanation , and execution..

With best regards,
Ravi teja.

pinealservo, 2018-06-19:
I'm afraid I don't know of anyone who's done any experimenting with the video functionality in the avtp_pipeline framework recently.  I'm fairly sure it's worked in the past and should not be too far away from working condition, but gstreamer is a fairly complex beast in itself and there are a lot of reasons it could fail. There are some environment variables you can set that will tell the gstreamer libraries to emit different levels of debug output; I'd recommend looking for some general gstreamer debugging information to enable that and get an idea about where in the gstreamer processing chain things are malfunctioning.

Ravi-Teja-konda, 2018-07-13:
Hello Everyone,

After looking in to the gstreamer elements behaviour i changed the gstreamer pipeline for ==>"**intf_nv_gst_pipeline" as "appsrc name=avbsrc ! decodebin ! queue ! videoconvert ! videoscale ! autovideosink**" for Mpeg2ts_gst_listener for streaming .ts file 
==>change the intf_nv_gst_pipeline to "**appsrc name=avbsrc ! decodebin3 name=dec ! queue ! videoconvert ! videoscale ! autovideosink  dec. ! queue ! audioconvert ! audioresample ! autoaudiosin**k"
for both audio and video.
====For Camera===
==>on the listener side change to " **appsrc name=avbsrc ! decodebin ! queue ! videoconvert ! videoscale ! autovideosink**"
==>on the talker side = "**v4l2src ! video/x-raw,width=640,height=480,framerate=30/1 ! decodebin ! videoconvert ! videoscale ! x264enc ! queue ! h264parse ! mpegtsmux ! appsink name=avbsinkeo/x-raw,width=640,height=480,framerate=30/1 ! decodebin ! videoconvert ! videoscale ! x264enc ! queue ! h264parse ! mpegtsmux ! appsink name=avbsink**" and change configuration map_nv_item_count=200,sr_class=B,sr_rank=1, try to increase the deficit_usec=5000 as changing these parameters worked fine but with some errors....

make sure that you have installed all the dependencies of gstreamer.
I installed gstreamer gst-1.14.1 manually.
 
Thanks,
Ravi Teja.

Markus-E-Gruber, 2018-07-25:
Hello @Ravi-Teja-konda ,
thank you very much for the investigation! It worked for me :)
The .ts Video appears correctly on the Listener!
But i am confused with the IPG. I was hoping there are 8000 packets per second (125 usec) like in the audio streaming applications (wav_file_talker/alsa_listener and simple_talker/listener)

The IPG here seems a bit undefined and is way below the 8000 packets per seconds.
btw: I repeated the streamed .ts Video so the graph repeats periodically.
[image]

Has somebody a clue what is happening here? Is this the normal behavior with Videostreaming?
Thank you in advance!

Best Regards
Markus

MarcinMiklas, 2018-07-25:
Yes, it is normal behaviour for video streaming. You will never get constant frames per second with compressed video.

TengXu94, 2018-07-31:
Dear all,
I was also looking into IPG problem, and I would like to ask a question:
Since there are not guaranteed 8000frames/s for Stream A, the 2ms delay is still guaranteed or not even this requirement?

TengXu94, 2018-08-02:
Thank you very much for the answer!

shirArg, 2019-01-03:
Did someone managed to make avtp_pipeline work with video in the end? I want to make sure this is possible before trying. 

Any recommendations?

Thank you very much :)

athul7pk, 2019-01-03:
yeah ...I have tried without srp and mrp and it will work for sure

shirArg, 2019-01-06:
What about with SRP? :)

athul7pk, 2019-01-06:
Definitely it will work ...I was just suggesting to avoid complexity.  first you can try without additional features like srp and later you can enable this feature after avtp

shirArg, 2019-01-06:
Thanks! I'll try it! :)

shirArg, 2019-01-07:
Hi, 

I tried to run the ts file example with gst listener. Following this thread, I was able to make it work, using:

In the talker side: 
----------------------
sudo ./openavb_harness -I $IFNAME -s $STREAMS -d 0 -a a0:36:9f:2d:01:ad mpeg2ts_file_talker.ini,sr_class=$CLASS,map_nv_tx_rate=$RATE,max_transit_usec=$TRANSIT_USEC,report_seconds=$REPORT

I try to stream the first segment of the two first examples from here- https://www.w6rz.net/.

In the listener side: 
-------------------------
sudo ./openavb_harness -I $IFNAME -s $STREAMS -d 0 -a a0:36:9f:2d:01:ad mpeg2ts_gst_listener.ini,sr_class=$CLASS,map_nv_tx_rate=$RATE,max_transit_usec=$TRANSIT_USEC,report_seconds=$REPORT

In the configuration-
intf_nv_gst_pipeline = appsrc name=avbsrc ! decodebin ! queue ! videoconvert ! videoscale ! autovideosink
(as recommended here by @Ravi-Teja-konda )

In both sides I built everything with SRP support. 

Result
----------
 I can see the movie in the other computer, but it is very slow, and getting stuck many times. 

Two specific problems: 

1. In the talker, there are a lot of messages saying- 
`OPENAVB MPEG2TS Interface] ERROR: Media queue full` 
Anything I can do to solve this?

2. In the listener- I expected one video window to be opened, but there are 7 instead (one for each stream, I assume). Is that normal? 

I would appreciate any help with this, thanks!

JakubMielczarek81, 2019-01-08:
Hi,
The reason you are getting "Media queue full" errors is because the media queue item size is "#define MPEG2TS_MQITEM_SIZE                     9024". This is how many bytes are put during each iteration into media queue item. The number of MPEG2TS packets put during the same iteration into AVTP frame is 7, which is 7*192=1344 bytes, but 7*188=1316 are consumed from the media queue item.

shirArg, 2019-01-08:
Thank you for your reply. 

My rate is 8000. 
I also found out that there is a configuration of- 

# map_nv_item_count: The number of media queue elements to hold.
map_nv_item_count = 20

Which I thought might be related, but I don't know.

And- I actually tried it with another movie (mp4 downloaded from Youtube and converted to ts using [this](https://www.media.io/convert/mp4-to-ts.html) ) and it did work, so I thought it might be related to the movie itself too. 

What do you think?

JakubMielczarek81, 2019-01-08:
The rate is reasonable, unless you have a high bitrate stream (more than 10.5Mbit/s), but then the number of frames should be increased. Maybe it is stream related.

The number of media queue items could make a change.

There is a "pacing"  mechanism enabled by default in the mpeg2ts file talker (can be disabled by setting "intf_nv_enable_proper_bitrate_streaming=0"). This mechanism tries to match mpeg2ts streaming rate to the playback rate, and may also introduce some small "slowdown" of playback.

shirArg, 2019-01-08:
Thanks, I'll try it and let you know. 

Regarding the listaner side- do you have any idea why I see 7 gstreamer windows instead of one? I think it might be one of the reasons to the slowing down, and not sure if it should be this way or not... 

Thanks!

raja2410, 2019-01-11:
Hi,
I can see the MPEG2TS movie  file on my listener side without any issues. But in the talker side I am seeing the error "Media queue full". In my talker configuration max_interval_frame =1 and I seeing the error in both 4000 and 8000 frame rate. Is there any way I can resolve the error by changing the configuration parameter values.

JakubMielczarek81, 2019-01-11:
Please have a look at https://github.com/AVnu/OpenAvnu/issues/787#issuecomment-452337522

raja2410, 2019-01-15:
Hi JakubMielczarek81,
From your comment I understand that due to media queue item size is 9024 (#define MPEG2TS_MQITEM_SIZE 9024) we are seeing the "Media queue full error ". Can this error be ignored for MPEGTS case.

JakubMielczarek81, 2019-01-15:
Hi,

In normal conditions it can be ignored. But if there are issues with reading data from media queue, it could indicate an error.
If you know what is the size of MPEG2TS packet of your source (188 or 192) you can experiment with setting MPEG2TS_MQITEM_SIZE to 188*7 or 192*7 - same as max size of mpeg2ts payload put by mapping in one AVTP frame.

JakubMielczarek81, 2019-01-17:
Hi,
My suggestion to change the MPEG2TS_MQITEM_SIZE to 188x7 was for mpeg2ts file talker.

chuanqima, 2019-07-11:
Hi，Markus

> Hi @Ravi-Teja-konda ,
> sorry the gstreamer implementation does not work for me yet.
>
> I'm not able to to play the incoming IEEE1722 video stream with "mpeg2ts_gst_listener.ini" on the Listener. I can [...]

Hi,Markus,
    I would like to ask you if you have tried to build avtp_pipeline with make avtp_pipeline (that implements SRP and mrp) when AVB_FEATURE_ENDPOINT=1.
I found that when AVB_FEATURE_ENDPOINT=1, the command openavb_harness was built to implement the ALSA example "Talker:
Sudo ./openavb_harness -I enp1s0 wav_file_talker.ini
Listener:
Sudo ./openavb_harness -I enp4s0 alsa_listener.ini "
There are always the following problems:
Through wireshark, you can see that 1722 packets are transmitted on the AVB network. However, the wav audio file played on the talker cannot be heard on the listener.

If I use AVB_FEATURE_ENDPOINT=0, build the project, and then implement ALSA, everything will be fine. By the way, my test environment is directly connected to two PCs.
Is it missing from me which file has not been modified? For example the endpoint.ini file?

The last question is, if I have ported the AVB_FEATURE_ENDPOINT=1 compiled protocol stack to my development board, can I limit his SRP function by modifying the endpoint.ini file?
I am looking forward to your quick reply, thank you very much.
best regards.
chuanqiMa

chuanqima, 2019-07-11:
> AVB_FEATURE_ENDPOINT = 0用于禁用mrp和srp它将避免麻烦的复杂性。
> 
> 您的gstreamer插件安装中缺少MPEG传输流解析器，
> 提供mpegtsdemux。插件的插件是坏插件的一部分，如果我的内存是正确的。
> 
> sudo apt-get install gstreamer0.10-plugins-bad ..如果你有ubuntu和gstreamer-0
Hi，athul7pk
    Can I complete the example in ALSA by constructing avtp(AVB_FEATURE_ENDPOINT = 1) on two directly connected PCs? I tried it, but there is no sound on the listener side. Is there a packet transmission of IEEE 1722 protocol in wireshark? Do you know how to solve this problem? Is it necessary for me to modify the endpoint.ini file?
Looking forward to your reply. thank