Avnu/OpenAvnu #796: Bandwidth Reservation not working..

Ravi-Teja-konda, 2018-06-12:
Hello everyone,

I am testing  OpenAvnu-master on ubuntu 16.04 with kernel version [4.4.0-119-generic]. I had completed testing basic audio, video streams on this version. I started testing the bandwidth reservation, and used an AVB Talker to send Class A  wav file streaming to Listener. 

 AVB bandwidth is  6 mbps when streaming wav file.

After that, my non_avb_bandwith _generator application from talker side started  to send the broadcast packets to make the whole bandwidth to 100M. I thought that the broadcast packets will be dropped, and the Class A streaming will still work fine, but in this case, on the listener side AVB bandwidth reduced to 3Mbps with glitches in the audio quality on the listener side and after increasing the non-avb bandwidth some more AVB bandwidth got reduced to 1 Mbps 

This was the message on the listener side when testing the Bandwidth reservation:

AVTP sequence mismatch: expected: 133; got: 136; lost 3
AVTP sequence mismatch: expected: 172; got: 176; lost 4
AVTP sequence mismatch: expected: 192; got: 194; lost 2
AVTP sequence mismatch: expected: 233; got: 237; lost 4

After looking in to the code i came to know that bandwidth reservation api [igb_set_class_bandwidth2()] which is called by the function [setupHWQueue()] in openavb_qmgr.c and the macro [AVB_FEATURE_IGB] that needs to be set for bandwidth reservation. but i couldn't find the appropriate Makefile to set to. and got errors while setting it to 1.

[CMake Error at platform/Linux/CMakeLists.txt:296 (add_library):
  Cannot find source file:
  /home/OpenAvnu/OpenAvnu-master/lib/avtp_pipeline/platform/generic/openavb_igb.c
  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx]

Is the selected process right to make work bandwidth resevation work properly ? or any other ways...

Waiting for reply,

Thank you,
Ravi Teja.

Ravi-Teja-konda, 2018-06-14:
Can someone please help me to resolve the issue ... how to compile the  project to reserve  bandwidth...

Thanks,
Ravi teja.

pinealservo, 2018-06-19:
To enable the hardware traffic shaping, you need to use the PLATFORM_TOOLCHAIN environment variable to switch from "generic" to "x86_i210_linux"; this will enable using the special features of the igb driver in the repository.

With the generic toolchain, we rely on the kernel's traffic shaping functionality to do shaping. It should do a reasonably good job if it's got enough resources; have you enabled the shaper daemon?  See the "run_daemons.sh" and other scripts at the top level of the repository for examples of getting all the necessary processes going.

Ravi-Teja-konda, 2018-06-20:
Hello Levi Pearson,

Thank you very much for the reply, I will make changes  for compilation as mentioned.
For your information about shaper daemon I'm running the shaper daemon by running run_daemons.sh 

I've changed endpoint.ini file shaper port number to 15365 on both talker and listener side and changed FQTSS mode to = 4(AVB_SHAPER_SW - Credit-based shaping per-class (in software))
on talker side. 
Do I need to run shaper daemon manually and pass commands manually to run BW reservation successfully?

Ravi-Teja-konda, 2018-06-20:
Hello Levi Pearson,

As mentioned switching PLATFORM_TOOLCHAIN environment variable from  "generic" to "x86_i210_linux"; I made changes to /OpenAvnu-master/lib/avtp_pipeline/avtp_pipeline.mk by changing PLATFORM_TOOLCHAIN to x86_i210_linux it got compiled. But when i enable the AVB_FEATURE_GSTREAMER =1 in make file i  get this following error

[ 96%] Building C object platform/Linux/avb_host/CMakeFiles/openavb_host.dir/openavb_host.c.o
[ 97%] Linking C executable openavb_host
../../../libintf_h264_gst.a(openavb_intf_h264_gst.c.o): In function `openavbIntfH264RtpGstTxCB':
openavb_intf_h264_gst.c:(.text+0xd83): undefined reference to `gst_al_rtp_buffer_get_timestamp'
collect2: error: ld returned 1 exit status
platform/Linux/avb_host/CMakeFiles/openavb_host.dir/build.make:116: recipe for target 'platform/Linux/avb_host/openavb_host' failed
make[4]: *** [platform/Linux/avb_host/openavb_host] Error 1
CMakeFiles/Makefile2:1156: recipe for target 'platform/Linux/avb_host/CMakeFiles/openavb_host.dir/all' failed
make[3]: *** [platform/Linux/avb_host/CMakeFiles/openavb_host.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make[2]: *** [all] Error 2
avtp_pipeline.mk:9: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:135: recipe for target 'avtp_pipeline' failed
make: *** [avtp_pipeline] Error 2

Do i need to change the way i  complie or is it with the gstreamer package?

Waiting for reply

Regards,
Ravi Teja.

pinealservo, 2018-06-20:
Apparently an update in March added that function, but didn't record the dependency in the cmake files properly for this circumstance. I'll have to investigate what needs to change to fix it, but I believe the implementation code is there.

TengXu94, 2018-08-23:
I would like to apologise if I'm writing on a very old post.
I would like to ask a question:
since I have changed the PLATFORM_TOOLCHAIN environment variable from "generic" to "x86_i210_linux" and I have the shaper daemon running, I can assume that I have two "level" of traffic shaping, one at the kernel and one at the NIC?

Best,
Andrea

sofiamorseletto, 2019-03-15:
Hello everyone, I'm having the same issue as Ravi Teja. When I try to compile the project with AVB_FEATURE_GSTREAMER =1, I get the following error:

[ 96%] Building C object platform/Linux/avb_host/CMakeFiles/openavb_host.dir/openavb_host.c.o
[ 97%] Linking C executable openavb_host
../../../libintf_h264_gst.a(openavb_intf_h264_gst.c.o): In function `openavbIntfH264RtpGstTxCB':
openavb_intf_h264_gst.c:(.text+0xd83): undefined reference to `gst_al_rtp_buffer_get_timestamp'
collect2: error: ld returned 1 exit status
platform/Linux/avb_host/CMakeFiles/openavb_host.dir/build.make:116: recipe for target 'platform/Linux/avb_host/openavb_host' failed
make[4]: *** [platform/Linux/avb_host/openavb_host] Error 1
CMakeFiles/Makefile2:1138: recipe for target 'platform/Linux/avb_host/CMakeFiles/openavb_host.dir/all' failed
make[3]: *** [platform/Linux/avb_host/CMakeFiles/openavb_host.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make[2]: *** [all] Error 2
avtp_pipeline.mk:9: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:120: recipe for target 'avtp_pipeline' failed
make: *** [avtp_pipeline] Error 2

I'm also getting the following warnings (maybe this has something to do with the issue):

make[4]: Warning: File '/usr/include/glib-2.0/glib-object.h' has modification time 82052787 s in the future

Any news on this subject or any idea how I could solve this?

Any help is appreciated.
Thanks.
Best regards,
Sofia

JakubMielczarek81, 2019-03-15:
Hello @sofiamorseletto 

Maybe you are building with gstreamer 0.1. Try with 1.0.
The gstreamer al for 0.1 does not have gst_al_rtp_buffer_get_timestamp function.

Thanks,
Jakub

sofiamorseletto, 2019-03-19:
Thank you so much @JakubMielczarek81, that solved my problem.

Best,
Sofia

closed 2019-03-19 by andrew-elder

sofiamorseletto, 2019-06-21:
Hello, William. Sorry for the delayed response, these are the steps I followed to build with gstreamer1.0:

1. First I installed some dependencies that I was missing (maybe it's not necessary for you but I'll mention them just in case):

-  Doxygen

-  Graphviz

2. Then, I installed gstreamer 1.0 following this link: [gstreamer-installation](https://gstreamer.freedesktop.org/documentation/installing/on-linux.html?gi-language=c) 

3. Some dependencies will not be found so this script needs to be run:
```
list=$(apt-cache --names-only search ^gstreamer1.0-* | awk '{ print $1 }' | grep -v gstreamer1.0-hybris)
sudo apt-get install $list
```

4. Finally, another plugin needs to be installed:
`sudo apt install libgstreamer-plugins-base1.0-dev`

5. Recompile the project, changing the following parameters: 

- In _/.../OpenAvnu/lib/avtp\_pipeline/avtp\_pipeline.mk_:
` AVB_FEATURE_GSTREAMER ?= 1`

-  In _/.../OpenAvnu/lib/avtp-pipeline/platform/Linux/x86\_i210\_linux.cmake_: 
`set (GSTREAMER_1_0 1)`

and then running `sudo ./travis.sh` from the root directory of OpenAVnu

I hope this helps.

Best,
Sofia