Avnu/OpenAvnu #798: Difference between AVB_FEATURE_ENDPOINT=0 and AVB_FEATURE_ENDPOINT=1

Praveen-apur, 2018-06-16:
Hello,

I have a question regarding compiling OpenAvb-master. I'm hoping someone can help me with:

It is mentioned that in OpenAvnu/lib/avtp_pipeline/README.md for compiling : 
[Building AVTP pipeline without SRP or MAAP support
$ AVB_FEATURE_ENDPOINT=0 make avtp_pipeline] 
what does this mean? as we need SRP to run AVB. it was mentioned that with out SRP? can we run this project without SRP?
I have just entered the world of AVB. Any help will be highly appreciated. Thanks.

Regards,
Praveen-apur.

ahogen, 2018-06-18:
As far as I know, yes, you do need SRP (MSRP, MVRP) for an AVB endpoint. I don't believe there is another way to reserve bandwidth for an AVB stream properly without this standard.

Someone else can confirm this, but it may be that there is a software flag to disable SRP for testing/evaluation purposes. Not fir real-world useage.

Refer to [IEEE 802.1BA](https://ieeexplore.ieee.org/document/6032690/) for AVB endpoint requirements. Also, Jeff Koftinoff has a [nice FAQ](https://avb.statusbar.com/page/faq/) which may be very informative for you.

Praveen-apur, 2018-06-18:
Hello Ahogen,

Thank you very much for reply. I have one another question regarding bandwidth reservation 
file located in [Open-avnu/lib/avtp_pipeline/tl/CmakeLists.txt] which compiles according to the AVB_FEATURE_ENDPOINT and sets source files for endpoint. but, after checking all the *endpoint.c
files i came to know that none of the functions in the endpoint are calling function[igb_set_class_bandwidth2()] which is called by calling [setupHWQueue()]. actually *endpoint.c files are transmitting data through sockets.

These functions are only called by the files which are *_no_endpoint.c do you have any idea about this thing .like how to call these functions for AVB_FEATURE_ENPOINT=1 in order to reserve bandwidth?
 
Waiting for reply,
Praveen-apur.

Joezhaoz, 2018-06-18:
Hi Praveen,

AVB_FEATURE_ENDPOINT=0 means compiling avtp_pipeline to be static stream reservation, while set AVB_FEATURE_ENDPOINT=1 is to use SRP to do streaming reservation. SRP is not mandatory for automotive profile, so that's why static streaming reservation is designed. 

setupHWQueue() is called to set the CBS parameters on the I210, that's is the way to make sure the bandwidth is reserved for the specific stream.

Hope this will be helpful for you.

Praveen-apur, 2018-06-18:
Hello Joezhaoz,

Thank you very much for the quick reply, I have one question and clarification regarding bandwidth reservation 
-->AVB_FEATURE_ENDPOINT=0  is for static stream reservation for  expected traffic patterns which is  default functionality in automotive use cases right ?
-->AVB_FEATURE_ENDPOINT=1 It allows endpoints to dynamically reserve and release stream bandwidth

So, I want to check Dynamic Bandwidth reservation 
My question is when i compile with AVB_FEATURE_ENDPOINT=1 it should call SetupHWQueue() which is located in [openavb_qmgr.c] but checking function calls for AVB_FEATURE_ENDPOINT=1 no api is calling SetupHWQueue() it is getting called only in AVB_FEATURE_ENDPOINT=0 but it is not calling igb_set_class_bandwidth2 () because of conditional compilation flag [AVB_FEATURE_IGB] which needs to be set i tried to set that but i failed to set it. Do you have any idea where and in which file to set it ?
 
Any comments would be great help for me.

Waiting for reply,

Regards,
Praveen.

Praveen-apur, 2018-06-19:
Hello everyone,
  
Can some one please help me in understanding this project.

Thank you,
Praveen-apur

chuanqima, 2019-07-10:
hello,everyone
   I have a weird question and hope to get some help. I used AVB_FEATURE_ENDPOINT = 1 when compiling avtp_pipeline, but I commented out the value of preconfigured when configuring endpoint.ini. Can this achieve the effect of turning off SRP? 
  I have a problem now. On two directly connected PCs, if AVB_FEATURE_ENDPOINT = 0 is used, the compiled openavb_harness related command can be used to play wav format audio on the talker side and receive the sound of this audio file on the listener side. But if I use the openavb_harness command compiled with AVB_FEATURE_ENDPOINT = 1, I can't hear anything. why is it like this? Any help is very meaningful, thank you very much.