Avnu/OpenAvnu #799: OpenAVB building issue on generic platform

Praveen-apur, 2018-06-19:
Hello Everyone,

I'm using OpenAVB-master with intel_i210 NIC on ubuntu 16.04 with kernel version 4.4.0-119-generic. when i try to set the AVB_FEATURE_IGB flag in generic.cmake file I'm getting this following error 

CMake Error at platform/Linux/CMakeLists.txt:309 (add_library):
Cannot find source file:

/home/OpenAvnu-master/lib/avtp_pipeline/platform/generic/openavb_igb.c

Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp .hxx .in .txx
Call Stack (most recent call first):
CMakeLists.txt:37 (include)

Does this repo supports bandwidth reservation for generic version?

pinealservo, 2018-06-19:
Hi Praveen-apur,

If you want to make use of special hardware features of a NIC, you need to switch from the "generic" platform to one that matches your hardware, like the "x86_i210_linux" platform if you want to use the i210 features. You do this by changing the PLATFORM_TOOLCHAIN environment variable when you build. Switching to the x86_i210_linux platform should automatically enable the igb driver and the path will include the right files.

If you look at the lib/avtp_pipeline/avtp_pipeline.mk file, you'll see the variable and how it's used to select a specific toolchain cmake file to include, and you'll also see the directory where those are found where you can look at the options.  Note that "generic" doesn't have anything to do with the "generic" in your Ubuntu kernel version, it's just called that because it doesn't rely on any platform-specific features.

The default build now is a generic Linux one that makes no assumptions about special hardware features. The current shaper daemon will set up kernel-based traffic shaping that approximates Qav shaping well enough to do some simple audio streaming of one or two streams in an AVB network. There are now standard kernel interfaces to the Qav shaper, but we have not yet updated the shaper daemon to take advantage of them or the new software implementation of Qav.  If you would be interested in trying to implement that, we could probably help point you in the right direction.

Jtun53, 2019-07-11:
@pinealservo 

As a semi related question, should the same be done for avtp_avdecc.mk? I'm running on Ubuntu with i210 in pcie slot <-> MOTU device.