Avnu/OpenAvnu #802: Repo version of AVB?

Praveen-apur, 2018-07-05:
Hello ,

I'm looking for the release version this repo. but i found the older version.
Is the release updated ?

Regards,
Praveen.

andrew-elder, 2018-07-05:
Are you asking that the entire repo be tagged with an updated version number?

Praveen-apur, 2018-07-05:
Hello andrew,

I need repo version for the avb-master.

Thank you.

andrew-elder, 2018-07-05:
@Praveen-apur - there isn't a defined repo version at this time. I can create a tag on *master* if you like. Please let me know.

Praveen-apur, 2018-07-06:
Hello Andrew,

Thank you for the reply, creating the tag would be better for the master.
I would like to have a tag for the master.

Thank you,
Praveen.

pinealservo, 2018-07-06:
We have apparently tagged a few releases in the past when we had several more people contributing and doing testing, but I'm not sure there's much value in tagging releases right now, as we don't have the resources to do much validation beyond basic code review and the small number of automated unit tests run by the CI servers. The "master" branch updates are as much of a "release" as you are likely to get without more contributors.

Maybe tagging the merges from open-avb-next with the date on which they occur would be helpful and a bit more honest about what they represent than a full release tag.

andrew-elder, 2018-07-06:
@pinealservo - good point(s) regarding the tagging operation. I'll try to tag with the date over the weekend.