Avnu/OpenAvnu #804: Anything is available  in android domain  for AVB?

athul7pk, 2018-07-09:
yes of-course  its also a Linux ! .
but interaction to HAL is different.:(
so can any one kindly suggest a method for running AVB on android .
Thanks in advance..

pinealservo, 2018-07-09:
There could be some vendor-specific work related to Android-based IVI systems somewhere, but I don't know of any specifics. Most of the low-level work of managing the protocols and scheduling traffic would be identical to any other system based on a Linux kernel, but there would likely also be a Java layer on top to manage things. Anything remotely standardized there would have to be provided by Google or some automotive-oriented SoC vendor or industry alliance group.

It's possible that there are android-based IVI systems out there right now that just present AVB streams to Android as normal audio/video interfaces, but that's pure conjecture on my part; I haven't seen any code anywhere.

athul7pk, 2018-07-10:
thank you very much..

I am also not good in android,only have previous experience with Linux...