Avnu/OpenAvnu #806: Achieving Static Configuration in AVB?

Praveen-apur, 2018-07-11:
Hello everyone,

After resolving bandwidth reservation and traffic shaping issues I want to try the static bandwidth reservation.
From the discussions, I came to know that it can be done by making AVB_FEATURE_ENDPOINT=0 and PLATFORM_TOOLCHAIN="x86_i210_linux" for hardware traffic shaping and configure AVB switch for reserving bandwidth, configure talker,  listener.
Is this okay to achieve static bandwidth reservation?

Any comments would be helpful.

Thanks,
Praveen.

athul7pk, 2018-07-12:
Yeah
The static stream configuration is implemented by making the following changes:
->Stream reservation protocol (SRP) disabled
->MAAP disabled - static talker MAC address assignment
Set the source and sink stream information statically at startup of
the AVB application and then immediately enable the stream. Static stream information includes the
sink address, media clock, channel count and map, stream ID and VLAN number (if not simple talker you can give it in configuraion files .ini ). All of these optimizations can applied to the modified AVB Endpoint software using  AVB_FEATURE_ENDPOINT=0 .
Switch should be configured for the vlans :)

Praveen-apur, 2018-07-12:
Hello Athul,

Thank you for the reply. one last question 
If I am using the switch I should configure the switch for stream reservation, right?

Thanks,
Praveen.

athul7pk, 2018-07-12:
Yes ,At least you may need to configure vlan ports .
Configuration will be switch specific .
You can find the details in switch specification document .
for just confirming application is working or not ( without SRP ) you can enable broad cast in all port of the switch and you can try with one talker and one listener.