Avnu/OpenAvnu #809: Failed to open PTP clock device and PDelay Response Receipt Timeout

SaiSenthil, 2018-07-19:
I have observed below exception  while trying to run gptp on ubuntu 18.04 with i210 interface card. 
is there any other configuration is required before running gptp? 

open-avnu/build-openavnu/daemons/gptp$ ./gptp enp1s0 
INFO     : GPTP [15:17:30:876] gPTP starting
ERROR    : GPTP [15:17:30:876] Using clock device: /dev/ptp0
ERROR    : GPTP [15:17:30:876] Failed to open PTP clock device
ERROR    : GPTP [15:17:30:876] Failed to initialize hardware timestamper, falling back to software timestamping
STATUS   : GPTP [15:17:30:877] Starting PDelay
ERROR    : GPTP [15:17:30:877] bool LinuxNetworkInterface::getLinkSpeed(int, uint32_t*): Unknown/Unsupported Speed!
STATUS   : GPTP [15:17:33:877] *** Announce Timeout Expired - Becoming Master
STATUS   : GPTP [15:17:33:877] New Grandmaster "00:1B:21:FF:FE:DE:2C:FF" (previous "00:00:00:00:00:00:00:00")
EXCEPTION: GPTP [15:17:33:909] PDelay Response Receipt Timeout
EXCEPTION: GPTP [15:17:34:912] PDelay Response Receipt Timeout
EXCEPTION: GPTP [15:17:35:915] PDelay Response Receipt Timeout
EXCEPTION: GPTP [15:17:36:918] PDelay Response Receipt Timeout
EXCEPTION: GPTP [15:17:37:921] PDelay Response Receipt Timeout
EXCEPTION: GPTP [15:17:38:924] PDelay Response Receipt Timeout

Any help will be highly appreciated. Thanks.

Regards,
SaiSenthil

pinealservo, 2018-07-19:
The failure to initialize the hardware timestamper is suspicious. Are you really running with an i210, or are you in a virtual machine? The OS does not see the i210 when running in a VM.

If not using a VM, do you have any other NICs installed it might be getting confused with?

SaiSenthil, 2018-07-20:
Thanks Pearson for the quick response.

The interface  enp1s0 is for i210 card which has installed with igb_avb.ko module on 18.04 Linux desktop.
As per kernel message, it is confirmed that enp1s0 is for i210.

~~~
_[  511.336725] igb_avb 0000:01:00.0 enp1s0: igb: enp1s0 NIC Link is Up 10 Mbps Full Duplex, Flow Control: RX/TX
[  511.336828] IPv6: ADDRCONF(NETDEV_CHANGE): enp1s0: link becomes ready_
~~~

as per  below error:
_ERROR : GPTP [15:17:30:877] bool LinuxNetworkInterface::getLinkSpeed(int, uint32_t*): Unknown/Unsupported Speed!_

The link speed should be  >=100Mbps to avoid the above error. Why I am getting link speed is 10Mbps.

Any help will be highly appreciated. Thanks.

Regards,
SaiSenthil

SaiSenthil, 2018-07-20:
linkspeed issue is fixed and now I am getting only exception

$sudo ./gptp enp1s0
INFO     : GPTP [11:40:12:459] gPTP starting
ERROR    : GPTP [11:40:12:459] Using clock device: /dev/ptp0
STATUS   : GPTP [11:40:12:459] Starting PDelay
STATUS   : GPTP [11:40:12:459] Link Speed: 1000000 kb/sec
STATUS   : GPTP [11:40:15:459] *** Announce Timeout Expired - Becoming Master
STATUS   : GPTP [11:40:15:459] New Grandmaster "00:1B:21:FF:FE:DE:2C:FF" (previous "00:00:00:00:00:00:00:00")
EXCEPTION: GPTP [11:40:15:492] PDelay Response Receipt Timeout
EXCEPTION: GPTP [11:40:16:494] PDelay Response Receipt Timeout
EXCEPTION: GPTP [11:40:17:495] PDelay Response Receipt Timeout
EXCEPTION: GPTP [11:40:18:496] PDelay Response Receipt Timeout

SaiSenthil, 2018-07-20:
Nice, it is worked, able to get AsCapable: Enabled.

Thanks for the support.

pinealservo, 2018-07-21:
I'm glad to hear you got it working. What turned out to be the problem, in case someone else comes across the same situation?

wanhao2015, 2018-12-03:
I try to use this code on ubuntu 16.04,the other side is frdm_k64f board,it runs gptp demo of zephyr.they were wired together.I followed these procedures,but this problem occurred.Perhaps it's something wrong with the zephyr demo?
https://docs.zephyrproject.org/latest/samples/net/gptp/README.html

wanhao2015, 2018-12-03:
I'm sorry! I run the code of this website on ubuntu 16.04.
https://github.com/AVnu/gptp

pinealservo, 2018-12-04:
Can you get a wireshark trace of the network traffic on the Ubuntu machine? The most likely explanation is that it's not seeing any responses to its PDelay request; they could be missing or malformed or something else. Looking at the packets being sent and received is the easiest way to see what's going on.

raj-pandey, 2021-07-06:
@sselvama I am having the exact same issue. How did you get it fixed? 3 years long time, I hope the this thread is not dead.