Avnu/OpenAvnu #811: mjpeg video streaming issue..

Ravi-Teja-konda, 2018-07-23:
Hello everyone,

I am using openAVnu-master repo and i built the repo with the below flags:

AVB_FEATURE_ENDPOINT ?= 1 -- (SRP Enabled)
IGB_LAUNCHTIME_ENABLED ?= 0
AVB_FEATURE_GSTREAMER ?= 1 -- (Gstreamer Enabled)
PLATFORM_TOOLCHAIN ?= generic -- (Generic Platform)

I've tested mpeg2ts video streaming with SRP and MAAP enabled .it worked fine but with 2 seconds delay. But when i Started running mjpeg and also h264 video streaming talker is opening the camera and sending the data but on listener side it is not receiving  the AVTP packets. i tested pipelining gstreamer data to a file  just to check the size of file if it receives any packets but it has not worked..

Below is the log of mpeg_listener..

Opening: mjpeg_gst_listener.ini,ifname=enp3s0
Configuring: mjpeg_gst_listener.ini,ifname=enp3s0
Starting: mjpeg_gst_listener.ini,ifname=enp3s0
[1532166862:587902403 OPENAVB osalTime] INFO: local_time = 1532166862410810556
[1532166862:587912350 OPENAVB osalTime] INFO: ml_phoffset = -162534272798150, ls_phoffset = 141806892
[1532166862:587924561 OPENAVB osalTime] INFO: ml_freqffset = 1.000003, ls_freqoffset = 0.999952
[1532166862:587928003 OPENAVB Endpoint] INFO: AVTP Pipeline: 0.1.3 (Development)
[1532166862:590779632 OPENAVB Raw Socket] ERROR: Checking interface; ioctl(SIOCGIFFLAGS) failed: No such device
[1532166862:590794119 OPENAVB Endpoint] ERROR: Invalid value: section=network, name=ifname, value=eth0
[1532166862:590820200 OPENAVB Endpoint] ERROR: Error in INI file: endpoint.ini, line 4
[1532166862:590860154 OPENAVB Talker / Listener] INFO: AVTP Pipeline: 0.1.3 (Development)
[1532166862:609502809 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbMapMjpegInitialize
[1532166862:609521411 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbIntfMjpegGstInitialize
[1532166862:611063189 OPENAVB Endpoint SRP] INFO: detected domain Class A PRIO=3 VID=0002...
[1532166862:611078892 OPENAVB Endpoint SRP] INFO: detected domain Class B PRIO=2 VID=0002...
[1532166862:611960827 OPENAVB Listener] INFO: Attach a0:36:9f:25:36:40/1
[1532166862:716680821 OPENAVB Endpoint] INFO: SRP listener callback uid=1: tlkrDecl=1
[1532166862:716936793 OPENAVB Endpoint] INFO: SRP listener callback uid=1: tlkrDecl=1
[1532166862:717600698 OPENAVB Listener] INFO: Starting stream: a0:36:9f:25:36:40/1
[1532166862:719271973 OPENAVB MJPEG Interface] INFO: Pipeline: appsrc name=avbsrc ! filesink location= test.mjpeg
[1532166862:719538180 OPENAVB Raw Socket] INFO: Using *pcap* implementation
[1532166863:036997756 OPENAVB Endpoint] INFO: SRP listener callback uid=1: tlkrDecl=1
[1532166863:037176160 OPENAVB Endpoint] INFO: SRP listener callback uid=1: tlkrDecl=1
[1532166864:779905745 OPENAVB Endpoint MAAP] INFO: MAAP address range 5 acquired:  0x91e0f000f097-0x91e0f000f0b6 (Size 32)
[1532166866:911931709 OPENAVB Endpoint] INFO: SRP listener callback uid=1: tlkrDecl=1
[1532166868:946701911 OPENAVB Endpoint] INFO: SRP listener callback uid=1: tlkrDecl=0
[1532166869:750542644 OPENAVB Listener] INFO: Stopping stream: a0:36:9f:25:36:40/1
[1532166869:750551695 OPENAVB Listener] INFO: RX a0:36:9f:25:36:40/1, Totals: calls=70, frames=0, lost=18446744073709551615, bytes=0

Any comments would be a grate help.

Thank you,
Ravi Teja.

Ravi-Teja-konda, 2018-07-23:
Hello JXter,

Thank you for the reply, I've tried setting interface name in endpoint.ini but no result.

JXter, 2018-07-24:
Are you still getting this error,
`ERROR: Error in INI file: endpoint.ini, line 4`
Setting the value in endpoint.ini in bin directory will solve this error temporary; or if you want a permanent solution, change it in endpoint.ini in endpoint directory and compile the pipeline again.

Ravi-Teja-konda, 2018-07-24:
Hello JXter,

ERROR: Error in INI file: endpoint.ini, line 4
got resolved , but the video displaying issue  for mjpeg is not yet resolved..

JXter, 2018-07-24:
Can you share the new logs. I don't see any other error in the shared logs.

Ravi-Teja-konda, 2018-07-24:
Hello JXter,

Thank you for the quick response. The log shows no error but I'm missing the packets  which are sent from the talker. it is like [INFO: RX a0:36:9f:25:36:40/1, Totals: calls=70, frames=0, lost=18446744073709551615, bytes=0 ] listener detects the domain class and attaches to the talker stream and runs the gstreamer pipeline with no error but it is unable to get the AVTP Packets from the talker ..

I've checked talker by pcap it is sending the AVTP packets..