Avnu/OpenAvnu #812: Difference between PCAP and IGB implementation

Ravi-Teja-konda, 2018-07-23:
Hello Everyone,

I just wanted to know difference between the implementation of PCAP and IGB.. which is used on the listener side..
Is the flag in the avtp_pipeline [IGB_LAUNCHTIME_ENABLED ] is related to the IGB ,PCAP implementation?

Waiting for reply,

Thanks in advance.
Ravi Teja.

pinealservo, 2018-07-23:
The avtp_pipeline framework defines an API for accessing a NIC; the API is called 'rawsock' in the code. There are a lot of implementations available, not just the ones based on pcap or the igb library. You can choose one based on the standard sendmsg Posix API or a memory-mapped interface like the one libpcap uses, but without libpcap itself. Choosing one that's different from the default will involve digging a bit more deeply into the cmake build of avtp_pipeline; I can't remember all the details off the top of my head.

In addition to the choice of rawsock implementation, the IGB_LAUNCHTIME_ENABLED flag enables a macro used to conditionally compile some code needed for using the scheduled launch functionality available to igb. You can see all the places it appears by typing "IGB_LAUNCHTIME_ENABLED" in the GitHub search bar and asking it to search just in this repo.

The IGB_LAUNCHTIME_ENABLED should really just be LAUNCHTIME_ENABLED; the newer Freescale/NXP SoCs have a launchtime feature as well, and there's an in-progress kernel interface to using these launch time features. The most significant bit of code that the flag enables is in openavb_avtp.c, where unaltered media item times are pulled from the media queue for use by the rawsock implementation (see the call to openavbRawsockTxFrameReady), of which only the igb_rawsock implementation currently does anything useful with.

One other significant difference is in openavb_talker.c; when you do not set the IGB_LAUNCHTIME_ENABLED flag, the main tx loop will attempt to evenly space transmissions by busy-waiting until the start of each transmit interval (unless the spin_wait flag is not set, in which case it will try to ask the kernel to wake it at appropriate intervals either way). The busy-waiting is thus disabled when you are using the launch time feature, since it has the effect of making the frames wait in the NIC egress queue.

sofiamorseletto, 2019-06-21:
Hello, everyone. 

Thank you @pinealservo for the thorough explanation. 

What would you say are the main advantages and disadvantages of each implementation (pcap and igb)?

My hardware configuration consists on a couple of Netgear GS716Tv3 switches that work as bridges between several devices. Some devices work as talkers and the other as listeners. And there are three traffic streams: class A, class B and Best Effort traffic. The devices have Ubuntu 16.04 and the igb_avb driver installed and an Intel® I210 1Gbit/s NIC. 

What do you think would be the most appropriate implementation for my setup?

Any help is appreciated.
Thank you.
Best regards,
Sofia

andrew-elder, 2019-06-21:
Anything that supports packet launchtime in both the hardware and the software stack is the better implementation. I see you are using Intel I210 NICs, so you can use the launchtime interface right from the start.

Using the packet launchtime feature lets you met the AVB traffic shaping requirements with 100% accuracy.

pinealservo, 2019-06-21:
If you're using the i210 NIC, then the igb implementation with IGB_LAUNCHTIME_ENABLED is the obvious choice with the current avtp_pipeline. However, even better would be use of the standard Linux APIs if you have a new enough kernel, which will probably involve a new rawsock implementation or maybe a tweak of an existing one like the one that uses sendmmsg. In any case, a production application is going to require pretty intimate knowledge (followed by tweaking/extending) of the existing avtp_pipeline code, and you'll be best served by experimenting with the different options.