Avnu/OpenAvnu #818: ERROR    : GPTP [20:00:17:754] Received Follow up but there is no valid link delay

hrdksny, 2018-08-02:
Hi
When I am running gPTP at both machine, I am getting below error at listener side.
ERROR    : GPTP [20:00:17:754] Received Follow up but there is no valid link delay.
On further investigation. I am curious about why "ONE_WAY_DELAY_DEFAULT = INVALID_LINKDELAY" is set in the common_port.hpp ?
Dose ONE_WAY_DELAY_DEFAULT value should be platform specific ? or Above configuration is fine ?
As due to above condition (!port->getLiknDelay()) always been true and observing above error.
How can I get rid of above gPTP error ?
Should I change the ONE_WAY_DELAY_DEFAULT value as per one way ethernet cable delay ?
My setup is x86 machine connected directly through Ethernet cable.

andrew-elder, 2018-08-02:
Maybe you can increase the log level to see what is going on.

hrdksny, 2018-08-02:
After debugging . . I have changed  ONE_WAY_DELAY_DEFAULT value to 200 and neighborPropDelayThresh as 1300. So I can't get the error. Is this ok ?
Can I change those values to use gPTP daemon ?

hrdksny, 2018-08-02:
No

andrew-elder, 2018-08-02:
Do both machines support hardware timestamping of 802.1AS packets?

hrdksny, 2018-08-02:
yes

Shuchish, 2021-12-07:
Getting the same issue, changed ONE_WAY_DELAY_DEFAULT value to 200 and neighborPropDelayThresh as 1300 also but still getting the same results