Avnu/OpenAvnu #821: RFC net-next Qbv implementation with taprio qdisc

Markus-E-Gruber, 2018-08-10:
Hello erverybody,
following patchset was released in July, and it should implement Qbv in the Linux Kernel as a qdisc but unfortunately i have some issues testing it.
https://patchwork.ozlabs.org/cover/943843/
This is an updated version of the Qbv implementation discussed in this issue #686 .

I used the 4.18.rc6 Kernel of David Millers net-next Tree and patched it with these two patches:
https://patchwork.ozlabs.org/cover/943843/
https://patchwork.ozlabs.org/cover/938991/

I also built and installed the patched iproute2 version that is provided here:
https://github.com/vcgomes/iproute2/tree/taprio-RFC-v1

Unfortunately adding the new qdisc "taprio" throws an Error:

```
$ tc qdisc add dev enp2s0 parent root handle 100 taprio \
	    num_tc 3 \
	    map 2 2 1 0 2 2 2 2 2 2 2 2 2 2 2 2 \
	    queues 1@0 1@1 2@2 \
	    sched-file ~/gates.sched \
	    base-time 1528743495910289987 \
	    clockid CLOCK_TAI
**RETNETLINK answers: Invalid argument**
```
I created a gates.sched file with following content:
```
S 01 300000
S 02 300000
S 04 300000
```
Have some body a clue what is the invalid Argument?
Has somebody already tested this?

Best Regards
Markus

Ravi-Teja-konda, 2018-08-28:
Hello Markus,

I'm Feeling good that you have stepped in to TSN.

Sorry for posting wrong question here, but I need your help in understanding the TSN.

I would like to know the hardware dependencies  for both endpoint and bridge in TSN network.
any comments would be a great help.

Thanks,
Ravi Teja.

ghindman, 2018-08-28:
Hello Markus,

This series came from my team here at Intel, but those folks are mostly on the kernel facing side - is there a reason you're asking here rather than on the netdev mailing list?  There have been a fair number of folks on the kernel side trying out these patches so you might have better luck there.  Let me know if that doesn't work for you, for some reason.

Additionally, we have a v1 taprio patchset pending based on the community feedback from the RFC you're currently using - we were just waiting for the next merge window to open up, which should be in a week or so.  You might want to wait for that, unless you have immediate need. (the RFC does work as is, though)

andrew-elder, 2018-08-28:
@ghindman - that's exciting news! Are you able to share a link to the patchset? I don't follow netdev....

ghindman, 2018-08-28:
Hello @andrew-elder - the patchwork patches above for taprio are still the latest, and the ETF patches are now in the 4.19 tree proper.  I believe the iproute2 patch is also merged into the latest version.  We should have an updated version of taprio for consideration to merge shortly, now that the 4.20 merge window is open.  I can ping the thread back when posted, though I would encourage anyone interested in this work to start tracking netdev - our initial work is on i210, but there are other NIC vendors inquiring about  the interfaces - most of it is  hardware agnostic - so you'll likely see a handful of other TSN-capable devices using the same interfaces over the next year or so.  There are also software-fallback methods for devices without hardware offload support and looser latency requirements.

ghindman, 2018-08-28:
FYI the pending v1 taprio patches are here: https://github.com/vcgomes/net-next/tree/taprio-v1 though fair warning that they've not been thoroughly tested after reworking some of the clocking.  That will likely happen next week ahead of upstreaming.

vcgomes, 2018-09-28:
Sorry for the delay, I just updated the branch at https://github.com/vcgomes/net-next/tree/taprio-v1 with what we want to propose upstream today.

maniskul, 2018-10-09:
@vcgomes Hi Vincent, could you please share the results as well that you have obtained after testing the taprio (source link: https://gist.github.com/jeez/bd3afeff081ba64a695008dd8215866f). I have gotten some results but not sure if they are correct as I don't have any reference results at the moment to compare with. 
Thanks,
Manish