Avnu/OpenAvnu #830: SR Class Priority Mismatch

TengXu94, 2018-09-11:
Dear all,
I am seeing a weird problem, my setting consist on:
 
PC1 [NIC: I210] <--> Motu AVB Switch <--> [NIC: I210] PC2

In which the switch is the gPTP Master Clock.
When I try to stream either audio or video between the two PC's using the respectively configuration files, the SR Protocol does not go well.
With Wireshark, is possible to see from Talker Side , Talker Advertise frames, in the other hand in the Listener Side, Talker Failures frames only can be seen , with failure code 19 (SR Class Priority Mismatch), even if on the configuration files are configured to be the same.
I run all OpenAvnu daemons with "run_daemons".

If anyone has some idea of this problem, would be really appreciated!
Best

closed 2018-09-11 by TengXu94

pinealservo, 2018-09-11:
I guess you figured this out? Was it a configuration problem? Was the Motu switch firmware up-to-date? There are a lot of things that can go wrong, your problem + solution might help someone else!

TengXu94, 2018-09-11:
Yes, it was my bad.
In Wireshark, I have seen that both PC's "Domain" declarations were corrupted, that's because, 
after reading #335 (that was for an end-to-end connection), I made the endpoints to declare first the domain with `mrp_register_domain` before getting any domain attribute, if any, with `mrp_get_domain`. 
I changed it back to normal, checked the domain declarations with `examples/mrpValidate` and double checked with Wireshark that the Listener was getting Talker Advertise instead Talker Failure and it was fine.

Still I don't get, for video stream (mpeg2ts), sometimes I get 1 `avtp sequence mismatch`, rarely more, even if with Wireshark I can count the same number of frames sent from Talker are received from the Listener.